ghdl/ghdl #2197: GHDL fails to build on Windows (MINGW64) with LLVM 15.

cr1901, 2022-09-24:
**Description**
I build GHDL once per week to keep up to date with FPGA-related tooling that I develop on (including GHDL). This week's build failed with an internal error, and it appears to be related to LLVM 15 (but I cannot confirm for certain).

**Expected behaviour**
GHDL successfully builds.

**How to reproduce?**

```sh
mkdir build-ghdl
cd build-ghdl
../ghdl/configure --prefix=$PREFIX --enable-libghdl --with-llvm-config
make
```

I eventually got `GHDL Bug occurred` below quite a bit into the build.

**Context**

- OS:
   ```
   William@DESKTOP-3H1DSBV MINGW64 ~/Projects/FPGA/ghdl
   $ uname -a
   MINGW64_NT-10.0-19044 DESKTOP-3H1DSBV 3.3.6-341.x86_64 2022-09-20 22:07 UTC x86_64 Msys
   ```
- Origin:
  - [x] Built from sources: `73419afb3`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (0.37.0.r3149.g73419afb3.dirty) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-w64-mingw32
C:\msys64\home\William\Projects\FPGA\build-ghdl\lib\ghdl\std\v87\
Command line:
C:\msys64\home\William\Projects\FPGA\build-ghdl\ghdl1-llvm.exe --std=87 --bootstrap --work=std -c -o textio.o ../../src/std/v87/textio
.vhdl
Exception PROGRAM_ERROR raised
Exception information:
raised PROGRAM_ERROR : EXCEPTION_ACCESS_VIOLATION
Load address: 0x7ff642140000
Call stack traceback locations:
0x7ff6424939b8 0x7ffa15b7235d 0x7ffa15b214a2 0x7ffa15b70e8c 0x7ff990e4f6c4 0x7ff642148108 0x7ff6424072f3 0x7ff642406c06 0x7ff642406cad
 0x7ff6423ef3e2 0x7ff642453dec 0x7ff64245581c 0x7ff642456f63 0x7ff642142cee 0x7ff6421413ac 0x7ff6421414e4 0x7ffa146d7032 0x7ffa15b2269
f
******************************************************************
```

**Additional context**

I did `pacman -Syuu` before the build, as is customary for me to do each week. This appears to have upgraded me to LLVM 15.

```William@DESKTOP-3H1DSBV MINGW64 ~/Projects/FPGA/ghdl
$ git reflog
73419afb3 (HEAD -> master, origin/master, origin/HEAD) HEAD@{0}: pull origin master: Fast-forward
e5de56d02 HEAD@{1}: pull origin master: Fast-forward
```

Trying to build with my second most-recent checkout (`e5de56d02`) gives me a `Unhandled version llvm 15.0.0` error.

tgingold, 2022-09-24:
Yes, I am aware of that.  LLVM 15 has changed the API a lot, so there is work to be done.

cr1901, 2022-09-24:
Sometimes when I upgrade to the bleeding edge, I get what I deserve :P. Unfortunately, it's not convenient to downgrade, so on Windoze I think I just won't have GHDL for now.

Right now, I don't have access to a Linux system with LLVM 15 (although I could build one), so I cannot confirm the problem is there as well.

tgingold, 2022-09-24:
You can still use windows, but stick to llvm 14 (or use mcode backend).

nickg, 2022-09-24:
You can probably get LLVM 15 to work just by adding a call to `LLVMContextSetOpaquePointers(/*context*/, false);` which restores the old behaviour.

https://llvm.org/docs/OpaquePointers.html#transition-state

tgingold, 2022-09-25:
Sure, but for llvm 16 the compatibility will be removed!

Paebbels, 2022-09-28:
I don't see a way to install LLVM 14 and 15 in parallel or to downgrade LLVM to 14 in MSYS2. According to pacman, there is only LLVM 15 in the package manager. Thus all MSYS2 installations that update packages from time to time via `pacman -Suyy` are affected by this braking change.

tgingold, 2022-09-30:
Now fixed, llvm15 is supported.

closed 2022-09-30 by tgingold