ghdl/ghdl #2201: GHDL crash:  resolution function of array of array of integer

sckoarn, 2022-09-27:
**Description**
While creating unit tests, ghdl has a problem with this code.  Did not confirm on Aldec as the packages have to be ??  in separate files.

**Expected behaviour**
This test should pass,  assert failure is testing content of signals.
from unc_subt2.proc1 @ 100: Test passed ...

**How to reproduce?**
Attempt to run the code below:

```-------------------------------------------------
--  declare signal with custom resolution funciton.
--  The signal is a vector of vector of integers
--  Apply the resolution function of integer element
--  to the array of array.

package composites is
  --  integer
  type mx_ivec is array (natural range <>) of integer;
  type mx_ivec_p is access mx_ivec;
  type mx_ivec_plst is array (natural range <>) of mx_ivec_p;
  type mx_ivec_plst_p is access mx_ivec_plst;
  type mx_ivec_arr is array (natural range <>) of mx_ivec;
end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.composites.all;
package reso_pkg is
  subtype rint is integer;
  type rint_vec IS ARRAY ( NATURAL RANGE <> ) OF rint;
  function areso (s : rint_vec) return rint;
  subtype res_rint is areso rint;
  subtype resob is (areso) mx_ivec;
  subtype resoba is ((areso)) mx_ivec_arr;
end package;

package body reso_pkg is
  function areso(s : rint_vec) return rint is
    variable rtn : rint := 0;
  begin
    IF (s'LENGTH = 1) THEN
      RETURN s(s'LOW);
    ELSE
      --report "resolution function passed the following items.";
      FOR i IN s'RANGE LOOP
        --report integer'image(s(i));
        if s(i) = rint'left then
          if rtn < 0 then
            rtn := rint'left;
          else
            rtn := rtn + s(i);
          end if;
        elsif s(i) = rint'right then
          if rtn > 0 then
            rtn := rint'right;
          else
            rtn := rtn + s(i);
          end if;
        else
          rtn := rtn + s(i);
        end if;
        
      END LOOP;
    END IF;
    --report "Returning: " & integer'image(rtn);
    RETURN rtn;
  end function areso;
end package body;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.composites.all;
use work.reso_pkg.all;

entity unc_subt2 is
begin
  process
  begin
    report "Test unc_subt2 starting: signal resolution.";
    wait;
  end process;
end entity unc_subt2;

architecture rtl of unc_subt2 is

  signal bv : bit_vector(7 downto 0) := "11000100";
  signal b : bit;
  signal clk : bit;
  signal tcnt : integer;
  
  signal a1 : res_rint;
  
  signal a11 : rint;
  signal a12 : rint;
  signal a13 : rint;
  
  signal arr1 : mx_ivec(7 downto 0);
  signal arr2 : mx_ivec(7 downto 0);
  signal arr3 : mx_ivec(7 downto 0);
  signal arr_r : resob(7 downto 0);
  
  signal arr_arr1 : mx_ivec_arr(3 downto 0)(7 downto 0);
  signal arr_arr2 : mx_ivec_arr(3 downto 0)(7 downto 0);
  signal arr_arr3 : mx_ivec_arr(3 downto 0)(7 downto 0);
  signal arra_r : resoba(3 downto 0)(7 downto 0);

begin
  clock: process
  begin
    wait for 1 ns;
    clk <= not clk;
  end process;

  proc1: process(clk)
    variable cnt : integer := 0;
  begin
    if clk = '1' then
      b <= '0';
      cnt := cnt + 1;
    else
      b <= not b;
      if cnt >= 50 then
        report "Test passed ..." severity failure;
      end if;
    end if;
    tcnt  <= cnt;
  end process;
  
  proca11: process
    variable v1 : rint := 0;
  begin
    wait on clk until clk = '1';
    v1 := v1 + 1;
    if v1 > 10 then
      v1 := 0;
    end if;
    -- fill array of array
    for i in arr_arr1'range loop
      for j in arr_arr1(i)'range loop
        arr_arr1(i)(j) <= v1;
      end loop;
    end loop;
    for i in arr1'range loop
      arr1(i) <= v1;
    end loop;
    wait for 0 ps;
    arra_r <= arr_arr1;
    arr_r <= arr1;
    a11  <= v1;
    a1 <= v1;
  end process;
  
  proca12: process
    variable v1 : rint := -5;
  begin
    wait on clk until clk = '1';
    v1 := v1 + 1;
    if v1 > 5 then
      v1 := -5;
    end if;
    
    -- fill array of array
    for i in arr_arr2'range loop
      for j in arr_arr2(i)'range loop
        arr_arr2(i)(j) <= v1;
      end loop;
    end loop;
    
    for i in arr2'range loop
      arr2(i) <= v1;
    end loop;
    --for i in 
    wait for 0 ps;
    arra_r <= arr_arr2;
    arr_r <= arr2;
    a12 <= v1;
    a1 <= v1;
  end process;
  
  proca13: process
    variable v1 : rint := -10;
  begin
    wait on clk until clk = '1';
    v1 := v1 + 1;
    if v1 > 10 then
      v1 := -10;
    end if;
    -- fill array of array
    for i in arr_arr3'range loop
      for j in arr_arr3(i)'range loop
        arr_arr3(i)(j) <= v1;
      end loop;
    end loop;
    
    for i in arr3'range loop
      arr3(i) <= v1;
    end loop;
    wait for 0 ps;
    arra_r <= arr_arr3;
    arr_r <= arr3;
    a13 <= v1;
    a1 <= v1;
  end process;
  
  process(clk)
    variable tint : integer;
  begin
    if clk'event and clk = '0' then
      --report "a1 is: " & integer'image(a1);
      assert a1 = a11 + a12 + a13
        report "ERROR:  Resolved signal not the value expected."  severity failure;
      
      
      for i in arr_r'range loop
        tint := arr_r(i);
        --report integer'image(tint);
        assert tint = a11 + a12 + a13
          report "ERROR:  Resolved signal not the value expected."  severity failure;
      end loop;
      
      for i in arra_r'range loop
        for j in arra_r(i)'range loop
          tint := arra_r(i)(j);
          --report integer'image(tint);
          assert tint = a11 + a12 + a13
            report "ERROR:  Resolved signal not the value expected."  severity failure;
        end loop;
      end loop;
    end if;
    --report "" & integer'image(a1);
  end process;
  
end rtl;

```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```vh_tests/types_5_08/ghdl$ ghdl -r --std=08 unc_subt2

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r101.g420a1d3e) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/project/dsim/kcampbell/superp/test/dsim/vh_tests/types_5_08/ghdl/
Command line:
ghdl -r --std=08 unc_subt2
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans.adb:1468 access check failed
Call stack traceback locations:
0x5602064bd5d3 0x5602066d071e 0x5602066d110c 0x5602066d7996 0x5602066d0f08 0x5602066d7996 0x5602066d8081 0x560206687646 0x56020668e087 0x56020668fdb2 0x5602066791d2 0x5602066e7e7e 0x56020664f6f1 0x56020664c803 0x56020665ad6e 0x5602066ecc5f 0x56020662b0e4 0x560206562e60 0x5602066f05bf 0x560206317a71 0x7fd38290c099 0x5602063161d8 0xfffffffffffffffe
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

diogratia, 2022-09-28:
The crash message doesn't occur using the last available nightly build release (27 days old). Your version shows a 2.0 development release from March of 2021 by its commit SHA version portion.

There was a commit in early August of this year that may have changed the behavior.

Resolving numerical types doesn't appear to do anything useful having no ability to do inband signaling, lacking metavalues.

dalex78, 2026-08-17:
# Issue #2201 — Crash resolving a signal of array-of-array-of-integer

Analysis by an AI agent. Recommend (optionally merging the fix and) closing the ticket.

## High-level summary

A signal of an array-of-array type (`mx_ivec_arr`, array of `mx_ivec`, array of `integer`) with a custom resolution function applied via a resolved subtype (`subtype resoba is ((areso)) mx_ivec_arr;`) crashed at run time with an internal `CONSTRAINT_ERROR : trans.adb:1468 access check failed` when multiple drivers were resolved together.

## Investigation / current status: already fixed on master

Reproduced the exact repro from the report against current `master` (`--std=08`): the design analyzes cleanly and simulates to completion, reaching the test's own "Test passed ..." self-check message (the test intentionally ends the simulation via `report ... severity failure` once all of its resolved-signal assertions succeed — matching the issue's "Expected behaviour" section exactly). No crash, no internal error.

This must have been fixed as part of general improvements to array-of-array resolution-function handling since this was reported (originally against `2.0.0-dev`, 2022). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/gna/issue2201/` (the exact repro from the report) which analyzes and simulates the design, expects the simulation to end via the deliberate `severity failure` self-check, and asserts no `GHDL Bug occurred` crash happens first. This locks in the current, correct behavior as a regression guard.

commit 9f6bcc1b (2022-09-28): vhdl-sem: avoid a crash after error.  Fix #2201
closed 2026-08-19 by tgingold