ghdl/ghdl #2202: GHDL crash: configuration

amb5l, 2022-09-27:
I've written a simple model of an encrypted Xilinx primitive to allow me to simulate my design in GHDL, and a configuration statement to substitute this in when required. I've debugged the model using the Vivado simulator, which is happy with the configuration declaration, but GHDL crashes:
```
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r760.g8e56ee72b) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-w64-mingw32
C:\work\tyto2\build\tb_hdmi_tpg\
Command line:
D:\devtools\msys64\mingw64\bin\ghdl.EXE -a --work=work --std=08 -fsynopsys -Wno-hide -Wno-shared -PD:\devtools\msys64\mingw64\bin\../lib/ghdl/vendors/xilinx-vivado -frelaxed C:/work/tyto2/src/common/tyto_types_pkg.vhd C:/work/tyto2/src/common/tyto_utils_pkg.vhd C:/work/tyto2/src/common/basic/sync_reg.vhd C:/work/tyto2/src/common/video_out/video_mode.vhd C:/work/tyto2/src/common/video_out/xilinx_7series/video_out_clock.vhd C:/work/tyto2/src/common/video_out/video_out_timing.vhd C:/work/tyto2/src/common/video_out/video_out_test_pattern.vhd C:/work/tyto2/src/common/video_out/hdmi_tx_encoder.vhd C:/work/tyto2/src/common/video_out/vga_to_hdmi.vhd C:/work/tyto2/src/common/audio_io/xilinx_7series/audio_clock.vhd C:/work/tyto2/src/common/audio_io/audio_out_test_tone.vhd C:/work/tyto2/src/common/basic/xilinx_7series/serialiser_10to1_selectio.vhd C:/work/tyto2/src/designs/hdmi_tpg/hdmi_tpg.vhd C:/work/tyto2/src/common/tyto_sim_pkg.vhd C:/work/tyto2/src/common/video_out/model_tmds_cdr_des.vhd C:/work/tyto2/src/common/video_out/model_hdmi_decoder.vhd C:/work/tyto2/src/common/video_out/model_vga_sink.vhd C:/work/tyto2/src/common/basic/xilinx_7series/secureip.vhd C:/work/tyto2/src/designs/hdmi_tpg/tb_hdmi_tpg.vhd C:/work/tyto2/src/designs/hdmi_tpg/cfg_tb_hdmi_tpg_no_secure_ip.vhd
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : vhdl-nodes.adb:2492
Load address: 0x7ff737a00000
Call stack traceback locations:
0x7ff737e51716 0x7ff737ba364d 0x7ff737d19e1e 0x7ff737d1a4cf 0x7ff737d1a113 0x7ff737d18d00 0x7ff737d1ef5c 0x7ff737cf6819 0x7ff737d64339 0x7ff737c77187 0x7ff737e2af5b 0x7ff737a036d4 0x7ff737a013ac 0x7ff737a014e4 0x7ffaa0d454de 0x7ffaa0f64859
```
I'm using today's snapshot of GHDL with the mcode backend on Windows 11 / MSYS2.

To reproduce the error, run `make ghdl` in the `build/tb_hdmi_tpg` directory of https://github.com/amb5l/tyto2/commit/68c543fcab9eeb4162cb581eaff35bb14b275505. Note that I've precompiled Xilinx Vivado libraries (VHDL 2008).

tgingold, 2022-09-28:
The configuration is not correct and ghdl emits an error message just before crashing.
I fixed the crash, but of course the error remains.

amb5l, 2022-09-28:
Thankyou for the incredibly fast response. I will check out and test the fix on Windows when you commit it. I have seen errors reported in another simulator, so I expect that you are right that the configuration is  not correct. Strange that Vivado is OK with it. I do find configuration declarations mind bending!

amb5l, 2022-09-28:
Confirmed - the configuration declaration was simply wrong. Thanks again for your help.

closed 2022-09-30 by tgingold

tgingold, 2022-10-11:
You're welcome.  I would agree they aren't very intuitive to use.

commit 4747007b (2022-09-30): testsuite/gna: rename issue and close #2202