ghdl/ghdl #2203: ghdl-ls: add column ranges while displaying errors

Aniruddha-Deb, 2022-10-01:
**Is your feature request related to a problem? Please describe.**
ghdl-ls doesn't provide a column range while reporting a syntax error: this leads to editors not highlighting/underlining the complete error, and only showing the error if we go to a single column
[image]
[image]

**Describe the solution you'd like**
Add a range of columns highlighting an error: preferably the limits of the token that caused the error

**Describe alternatives you've considered**
Highlighting the entire line (and ignoring columns) would probably be easier, but a hackier fix.

**Additional context**
None

tgingold, 2022-10-03:
Which editor are you using ?

Aniruddha-Deb, 2022-10-03:
sorry, this is Neovim with the [coc.nvim](https://github.com/neoclide/coc.nvim) plugin configured for `ghdl-ls`
```json
"languageserver": {
  "ghdl-ls": {
    "command": "ghdl-ls",
    "filetypes": ["vhd", "vhdl"],
    "rootPatterns": ["hdl-prj.json"]
  }
}
```

and `hdl-prj.json` contains project-specific settings.

tgingold, 2022-10-05:
Columns are available but not columns range.  VS Code deals better than neovim with that.

But I agree this should be added.