ghdl/ghdl #2206: GHDL error: cannot find resource library "altera_mf"

gruvw, 2022-10-05:
**Description**

I cannot run ghdl while using the `altera_mf` library.

I have those lines in one of my vhdl source files:
```
LIBRARY altera_mf;
USE altera_mf.altera_mf_components.ALL;
```

When I build using `-P=altera/` after using `/usr/local/lib/ghdl/vendors/compile-altera.sh --altera --source "/home/gruvw/intelFPGA_lite/21.1/quartus/eda/sim_lib"`, I still get the following errors:
```
/home/gruvw/SynologyDrive/EPFL/BA3/CA1/ba3-ca1-labs/lab2-Memory/vhdl/ROM_Block.vhd:35:9:error: cannot find resource library "altera_mf"
LIBRARY altera_mf;
        ^
/home/gruvw/SynologyDrive/EPFL/BA3/CA1/ba3-ca1-labs/lab2-Memory/vhdl/ROM_Block.vhd:36:15:error: unit "altera_mf_components" not found in library "altera_mf"
USE altera_mf.altera_mf_components.ALL;
              ^
```

Could I please get some help on how to compile my files?

Also, I'm sorry if this is not the right place to ask, but I could not find another one.

- OS: Ubuntu 22.04.1 LTS x86_64

tgingold, 2022-10-05:
Doesn't the script create altera_mf directory (and not only altera) ?

gruvw, 2022-10-05:
I fixed my issue. It was not related to ghdl in the end.

closed 2022-10-05 by gruvw