ghdl/ghdl #2208: Default Location for Pre-Compiled Third Party Libraries

ttobsen, 2022-10-06:
Hi everyone,

is it possible to configure a search folder system-wide so that GHDL is looking for precomilied 3rd party libraries (e.g. Xilinx Unisim) in it.

The problem I'm facing is, that I have to specify either in VUnit or on cli the Unisims path, which breaks the workflow compard to Modelsim where the libraries are specified in the modelsim.ini.

Is there something possible currently or would this be a feature request?

Best regards,
Tobi

Paebbels, 2022-10-08:
The implementation of a system wide `modelsim.ini` has some drawbacks...
One problem is handling of multiple vendor tool versions on the same machine.

Anyhow, VUnit can take additional library paths, so it translates to `-P` parameters for GHDL. In addition, GHDL either accepts the specific library directory or directory containing a group of VHDL libraries. So in stead of specifying unisim, you can give the Vivado directory and it can find from there unisim, uniprim, unifast, ...

ttobsen, 2022-10-08:
Thanks for your reply. The problem is, that I need to specify the library path in my run.py. Bu this won't work on different machines were the libraires are located at different places.

I need a way to configure the library path independendly from the code in the repository. A VUnit environment variable would be fine, but I don't think that something in this way exists. :disappointed:

tgingold, 2022-10-08:
There is a system-wide folder where ghdl is looking for precompiled libraries.  That's also where the ieee library is.
Use ghdl --dispconfig to see the path.

ttobsen, 2022-10-08:
Thanks a lot! This is a working solution for me. :smile_cat:

closed 2022-10-08 by ttobsen

Paebbels, 2022-10-08:
As `run.py` is normal Python code, you can define a system-wide environment variable and read that variable from Python via `os.getenv('PRECOMPILE_LIBRARY_PATH')`.

ttobsen, 2022-10-09:
Thanks for this hint!