ghdl/ghdl #2211: When should static indices that are out of range get checked?   Analysis, Elaborate, Run time?

JimLewis, 2022-10-08:
**Description**
When should static indices that are out of range get checked?   Analysis, Elaborate, Run time?
If I write code that exercises a non-existent index, but I provide other code to guard it so 
that under the conditions of this test case, the code with the illegal, static bounds never 
gets run, then it should be ok right?

This is related to LRM issue https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues/275

**Expected behaviour**
I expect the following code to run.  

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: Test1.vhd
entity Test1 is
end entity;
architecture a of Test1 is
    signal s : bit_vector(1 to 5);
begin
  p: process is
  begin
    s <= (others => '0') ; 
    if s'right <= 10 and s'left >= 10 then
      s(10) <= '1';
    end if;
    report "finished" ;
    wait;
  end process;
end architecture;
```
```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a Test1.vhd
ghdl --elab-run Test1
```

On the other hand, if I swap out the static bounds with a variable, all is ok:
```vhd :file: Test2.vhd
entity Test2 is
end entity;
architecture a of Test2 is
    signal s : bit_vector(1 to 5);
begin
  p: process is
    variable V : integer ;
  begin
    V := 10 ; 
    s <= (others => '0') ; 
    if s'right <= V and s'left >= V then
      s(V) <= '1';
    end if;
    report "finished" ;
    wait;
  end process;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a Test2.vhd
ghdl --elab-run Test2
```

tgingold, 2022-10-09:
At run time.
But in the former case, it happens when drivers are created, because a driver for `s(10)` has to be created and this is an error.
In the latter case, drivers for `s` are created and this is not an error.

JimLewis, 2022-10-09:
WRT s(10), the LRM says:
> 14.7.2 Drivers
> Every signal assignment statement in a process statement defines a set of drivers for certain scalar signals.
There is a single driver for a given scalar signal S in a process statement, provided that there is at least one
signal assignment statement in that process statement and that the longest static prefix of the target signal of
that signal assignment statement denotes S or denotes a composite signal of which S is a subelement. Each
such signal assignment statement is said to be associated with that driver. Execution of a signal assignment
statement affects only the associated driver(s).

The LRM seems to say that a driver should only be created for s(10) when it is also a subelement of s.   In addition, at least in this section, it does not say that it is an error or even erroneous that s(10) exists.   In this case in particular, the code containing the reference to s(10) is never executed - unless s is modified to include s(10).

JimLewis, 2022-10-09:
The same text is in issue 84 of 1076-1987.   It looks like from their perspective, you are correct.  There they decided:

> VASG-ISAC Analysis and Rationale
The proposed resolution is rejected. This is an “elaboration-time” error.
The first sentence of the first paragraph of Section 9.2.1 on page 9-4 states that
Every signal assignment statement in a process statement defines a set of drivers for
certain scalar signals.
This means that every signal assignment statement has at least one driver associated with it
(regardless of whether the statement can/will ever be executed). Therefore, in the first example
above, a driver will always be created for S. If the generic is passed a value of 0, then an error
must be issued when the driver is created (i.e., at elaboration time).
This does not burden or restrict a designer, since the model could more naturally be written using
an IF generate statement.

JimLewis, 2022-10-09:
Currently Questa only generated warnings.   I wonder if this changed later.  

Does anyone have the ISAC resolutions for later standards - particularly in a single document so they can be searched easily?

tgingold, 2022-10-09:
Yes, the issues say all.  A set of drivers has to be determined, whether or not the assignment is executed.

JimLewis, 2022-10-09:
I noticed that driver creation did not talk about resolved signals - that actually comes later.   It makes me wonder if I assign to subelement of a resolved array signal, don't I drive all elements of that array.   I see text about it later wrt which signals are active, however, I don't find anything in drivers.   I even checked 2008 to make sure something did not get deleted.

tgingold, 2022-10-16:
LRM08 6.4.2.3:

If a subelement of a resolved signal of composite type has a driver in a given process, then every scalar subelement of that signal shall have a driver in the same process, and the collection of all of those drivers taken together constitute one source of the signal.

closed 2022-10-16 by tgingold

JimLewis, 2022-10-16:
Ok.  But what is a resolved signal?

Also 6.4.2.3 (2019):
If a resolution indication appears in the subtype indication in the declaration of a signal or in the declaration
of the subtype used to declare the signal, then each resolution function in the subtype is associated
correspondingly with the declared signal or with a subelement of the declared signal. Such a signal
subelement is called a resolved signal.

Hence, this would be of the form:
```vhdl
type MyArrayType is array (integer range <>) of MyType ; 
type MyArrayOfArrayType is array (natural range <>) of MyArrayType ; 
function MyResolution (A : MyArrayOfArrayType) return MyArrayType ; 

**subtype MyResolvedArrayType is MyResolution MyArrayType ;** 
```

Put this into use:
```vhdl
signal MyArray : MyResolvedArrayType(1 to 10) ;
signal MySlv : std_logic_vector(1 to 10) ; 
. . .

MyArrayProc : process
begin
  MyArray(1) <= ... ; 
  wait ; 
end process ; 

MySlvProc : process
begin
  MySlv(1) <='1'. ; 
  wait ; 
end process ; 

```
Hence, the process MyArrayProc above creates a driver on all elements of MyArray, however, MySlvProc only drives MySlv(1) - as this is not a resolved signal.

JimLewis, 2022-10-16:
That said, I don't think the paragraph defining resolved signal is written very well as it has been the source of confusion in the past.