ghdl/ghdl #2217: Can't resolve overload for procedure call

avelure, 2022-10-14:
**Description**
I'm not able to see how it has problems determining the overload in this case. Neither mentor nor nvc shows any warnings.

**Expected behaviour**
That it can determine the right overload.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
  type t_msg_id is (ONE, TWO);
  type t_msg_id_panel is array (t_msg_id'left to t_msg_id'right) of boolean;
  type t_quietness is (NON_QUIET, QUIET);
  procedure enable_log_msg(
    constant msg_id       : t_msg_id;
    variable msg_id_panel : inout t_msg_id_panel;
    constant msg          : string      := "";
    constant scope        : string      := "";
    quietness : t_quietness := NON_QUIET
  )is begin
  end procedure;

  procedure enable_log_msg(
    msg_id    : t_msg_id;
    msg       : string;
    quietness : t_quietness := NON_QUIET;
    scope     : string      := ""
  )is begin
  end procedure;
begin
process(all)
begin
  enable_log_msg(ONE, "Test test");
end process;
end architecture beh;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run test
```

```
ghdl -a --std=08 test_overload_2.vhd
test_overload_2.vhd:30:17: can't resolve overload for procedure call
  enable_log_msg(ONE, "Test test");
                ^
test_overload_2.vhd:30:17: possible interpretations are:
test_overload_2.vhd:20:13: procedure enable_log_msg [t_msg_id, string, t_quietness, string]
test_overload_2.vhd:11:13: procedure enable_log_msg [t_msg_id, t_msg_id_panel, string, string, t_quietness]
c:\prog\GHDL\nightly-mingw64-llvm\bin\ghdl.exe: compilation error
```

**Context**
- OS: Windows
- Origin: nightly 2ae368913

tgingold, 2022-10-14:
I think ghdl is right here.  The type of an aggregate is always determined by the context, here a procedure call.
But as the procedure is overloaded, it is not possible to determine id.

avelure, 2022-10-14:
I'm not sure I understand what is supposed to be the aggregate in the code? 

I can boil it further down. It seems the problem is with string literal and arrays.
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
  type t_msg_id_panel is array (natural range <>) of boolean;
  procedure enable_log_msg(
    variable msg_id_panel : t_msg_id_panel
  )is begin
  end procedure;

  procedure enable_log_msg(
    msg       : string
  )is begin
  end procedure;
begin
process(all)
begin
  enable_log_msg("Test test");
end process;
end architecture beh;
```
If I change the type of the array to be integer instead of boolean it is able to determine the right one. If the type is a custom type, it won't work again.

tgingold, 2022-10-14:
Sorry, I meant string literals instead of aggregate.
This issue is the same as std.textio.write which requires type qualification to distinguish string and bit_vector.

tgingold, 2022-10-16:
Yes, you're right.
There was a missing check for character types.  I am testing a fix.
Thank you for the issue.

commit 22bc0c76 (2022-10-16): vhdl-sem_expr.adb(is_string_type): check character type.

Fix #2217
closed 2022-10-16 by tgingold