ghdl/ghdl #2218: No element bounds for self-determined aggregate + bug

avelure, 2022-10-14:
**Description**
In the following case, I would assume it could determine the bounds through a `unsigned()` conversion. At least I get no warnings on mentor or nvc.

**Expected behaviour**
It can determine the bounds.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
  type t_unsigned_vector is array (natural range <>) of unsigned;
  type t_range_uns_vec is array (natural range <>) of t_unsigned_vector;
  procedure ptest(
    constant range_vec : in t_range_uns_vec) is
  begin
  end procedure;
begin
process(all)
  variable v_sig_long_min : unsigned(127 downto 0) := x"8F000000000000000000000000000000";
  variable v_sig_long_max : unsigned(127 downto 0) := x"8F000000000000000000000000000008";
  variable v_slv_long_min  : std_logic_vector(127 downto 0) := x"8F000000000000000000000000000000";
  variable v_slv_long_max  : std_logic_vector(127 downto 0) := x"8F000000000000000000000000000008";
begin
  ptest((0 => (v_sig_long_min, v_sig_long_max)));
  ptest((0 => (unsigned(v_slv_long_min), unsigned(v_slv_long_max))));
end process;
end architecture beh;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run test
```

```
ghdl -a --std=08 test_element_bounds.vhd
test_element_bounds.vhd:21:15: no element bounds for self-determined aggregate
  ptest((0 => (unsigned(v_slv_long_min), unsigned(v_slv_long_max))));
              ^
c:\prog\GHDL\nightly-mingw64-llvm\bin\ghdl.exe: compilation error
```

If line 8 is commented we can also provoke a bug

```
ghdl -a --std=08 test_element_bounds.vhd
test_element_bounds.vhd:9:55: no declaration for "t_unsigned_vector"
  type t_range_uns_vec is array (natural range <>) of t_unsigned_vector;
                                                      ^

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r811.g2ae368913) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-w64-mingw32
C:\proj\public\minimal\
Command line:
C:\prog\GHDL\nightly-mingw64-llvm\bin\ghdl1-llvm.exe --std=08 -PC:\prog\GHDL\nightly-mingw64-llvm\lib/ghdl\ieee\v08\ -PC:\prog\GHDL\nightly-mingw64-llvm\lib/ghdl\ -c -o test_element_bounds.o test_element_bounds.vhd
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : vhdl-nodes.adb:862
Load address: 0x7ff740090000
Call stack traceback locations:
0x7ff7404732f6 0x7ff74012eac3 0x7ff7402c8a52 0x7ff7402c8a09 0x7ff7402c8b36 0x7ff7402c8a09 0x7ff7402c8b36 0x7ff7402ae975 0x7ff7402aeba2 0x7ff7402bea86 0x7ff7402beee6 0x7ff7402dc4bf 0x7ff7402dc834 0x7ff7402dc8df 0x7ff7402dd46a 0x7ff7402dd552 0x7ff7402ddc35 0x7ff7402dde81 0x7ff7402ddfe5 0x7ff7402a63ff 0x7ff7402ad220 0x7ff740284a69 0x7ff7403a4cb6 0x7ff7403a6a3c 0x7ff7403a8183 0x7ff740092cee 0x7ff7400913ac 0x7ff7400914e4 0x7ffcd9717032 0x7ffcd9f8269f
******************************************************************
c:\prog\GHDL\nightly-mingw64-llvm\bin\ghdl.exe: compilation error
```

**Context**

- OS: Windows
- Origin: nightly 2ae368913

avelure, 2022-10-30:
I checked with 1cc85c578 and it seems only the crash was solved, is the inability to determine the bounds expected?

tgingold, 2022-10-31:
You're right.
There are a few issues about this restriction.  I will fix it.

avelure, 2023-01-17:
Should we reopen the issue so that fixing of the inability to determine the bounds is not forgotten, or do you have another way of keeping track of this?

tgingold, 2023-01-17:
But it is now working.  I will just add the test.

commit 49167cfb (2022-10-16): vhdl-sem_expr.adb: avoid crash after error on aggregate.  Fix #2218
closed 2023-01-17 by tgingold