ghdl/ghdl #2223: GHDL Analysis Crash on Pointers and Multidimensional arrays, Part 2

JimLewis, 2022-10-18:
**Description**
Continuation of issue #2166 

**Expected behaviour**
Code analyze successfully.   The long term goal is that instead of 
using integer_vector, this type is abstracted by a generic and the
actual type is passed in.   That type could be an array type or not
(integer_vector, integer, bit_vector are all candidates),
however, with only VHDL-2008, the type is private and we do not
know.   

**How to reproduce?**

```vhd :file: MemPkgReproducer6 .vhd
entity MemPkgReproducer6 is
end;
architecture behav of MemPkgReproducer6 is
  function InitMemoryBaseType(Size : integer) return integer_vector is
    
  begin
    return (1 to Size => 0);
  end InitMemoryBaseType;
  subtype MemoryBaseType is integer_vector ;
  type MemBlockType      is array (integer range <>) of MemoryBaseType ;
  type MemBlockPtrType   is access MemBlockType ;
begin
  process
    variable MemArr : MemBlockPtrType;
    variable BlockWidth, BaseWidth : natural;
  begin
    BlockWidth := 4;
    BaseWidth  := 32;
    MemArr := new MemBlockType'( 0 to BlockWidth - 1 => InitMemoryBaseType(BaseWidth));
    wait;
  end process;
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a MemPkgReproducer6 .vhd
```

Small variation using the constant INIT_MEM_BASE

```vhd :file: MemPkgReproducer7 .vhd
entity MemPkgReproducer7 is
end;

architecture behav of MemPkgReproducer7 is
  function InitMemoryBaseType(Size : integer) return integer_vector is
    
  begin
    return (1 to Size => 0);
  end InitMemoryBaseType;
  subtype MemoryBaseType is integer_vector ;
  type MemBlockType      is array (integer range <>) of MemoryBaseType ;
  type MemBlockPtrType   is access MemBlockType ;
  constant INIT_MEM_BASE : integer_vector := InitMemoryBaseType(32) ; 
begin
  process
    variable MemArr : MemBlockPtrType;
    variable BlockWidth, BaseWidth : natural;
  begin
    BlockWidth := 4;
    BaseWidth  := 32;
    MemArr := new MemBlockType'( 0 to BlockWidth - 1 => INIT_MEM_BASE);
    wait;
  end process;
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a MemPkgReproducer7 .vhd
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ x] Package manager: `GHDL 3.0.0-dev (2.0.0.r670.ga259712cc)`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
ghdl -a --std=08 -Wno-library -Wno-hide --work=default --workdir=C:/tools/sim_temp/GHDL-3.0.0-dev/de
fault/v08 -PC:/tools/sim_temp/GHDL-3.0.0-dev ../../SandBox/MemPkgReproducer6.vhd
Error: ../../SandBox/MemPkgReproducer6.vhd:20:32: no element bounds for self-determined aggregate
Error:     MemArr := new MemBlockType'( 0 to BlockWidth - 1 => InitMemoryBaseType(BaseWidth));
Error:                                ^
Error: C:\tools\msys64\mingw64\bin\ghdl.exe: compilation error
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

JimLewis, 2022-10-18:
@umarcor   Isn't it a bug?   Why feature request?

umarcor, 2022-10-18:
@JimLewis we don't claim to support all of VHDL 2008 yet. Therefore, issues such as this are on the limit between a supported feature that works wrong (a bug) or a detail of the feature that needs to be completed (not a bug).

I typically label the features which are incomplete per their specific type, and use the bug label for the issue that generate an explicit "bug occurred" block. Since there is no explicit bug here, I wait until Tristan replies. He might acknowledge the bug (typically he will fix it and close the issue straightaway) or he might make some ref to the LRM. The former case implies labeling as 'Bug' while the latter is 'LRM Reading'. Normally, I don't feel myself confident enough to decide that upfront, so I do it post-mortem.

commit ca958871 (2022-12-11): trans-chap7.adb: handle any expression for bounds.

Fix #2223
closed 2022-12-11 by tgingold