ghdl/ghdl #2228: *** GHDL Bug occurred ***

MarteGit, 2022-10-21:
**Description**
Inside this tar.gz file (https://www.gaisler.com/products/grlib/grlib-gpl-2022.2-b4274.tar.gz) there is the design "designs/noelv-generic". In the ".../designs/noelv-generic/" directory I did "make ghdl-run" and I get the Bug report. 

**Expected behaviour**
The "analyze and elaborate/include" operations looks be done without errors. But when I try "run" I get this message. It suppose run the simulation but crash in the launch moment. 

**How to reproduce?**
Get the file (https://www.gaisler.com/products/grlib/grlib-gpl-2022.2-b4274.tar.gz), unpack, go to the directory "/designs/noelv-generic/" and run "make ghdl-run"

I try with ghdl-mcode and ghdl-gcc version (The Bug report is from ghdl-gcc version, but with mcode mode happen the same).

**Context**
Please, provide the following information:

- OS: Linux Ubunto 22.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 10.3.0
Target: x86_64-linux-gnu
/home/cgriscv/WorkShare/grlib-gpl-2022.2-b4274/designs/noelv-generic/
Command line:
/usr/local/libexec/gcc/x86_64-pc-linux-gnu/10.4.0/ghdl1 --ghdl-fexplicit --ghdl-fsynopsys --ghdl--mb-comments --ghdl--warn-no-binding -O2 --ghdl--workdir=gnu/work --ghdl--work=work -Pgnu/ -Pgnu/grlib/ -Pgnu/techmap/ -Pgnu/eth/ -Pgnu/opencores/ -Pgnu/gaisler/ -Pgnu/work/ -P/usr/local/lib/ghdl/ieee/v93/ -P/usr/local/lib/ghdl/ --work=gaisler --workdir=gnu/gaisler/ -quiet -o gnu/gaisler/iunv.s ./../../lib/gaisler/noelv/core/iunv.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap4.adb:302
Call stack traceback locations:
0x7fea53af3542 0x8387d0 0x83c43e 0x80b8c0 0x80d5df 0x8569db 0x858184 0x8597f2 0x8581f1 0x8597f2 0x857de6 0x82f8a2 0x82bdbb 0x85e48b 0x7e8518 0x66f348 0xcb3fc1 0x66461f 0x67469e 0x85e7d9 0x662247 0x7fea535aed8e 0x7fea535aee3e 0x669e23 0xfffffffffffffffe
******************************************************************
Execution of /usr/local/libexec/gcc/x86_64-pc-linux-gnu/10.4.0/ghdl1 terminated by unhandled exception
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap4.adb:302
Call stack traceback locations:
0x7fea53af3542 0x8387d0 0x83c43e 0x80b8c0 0x80d5df 0x8569db 0x858184 0x8597f2 0x8581f1 0x8597f2 0x857de6 0x82f8a2 0x82bdbb 0x85e48b 0x7e8518 0x66f348 0xcb3fc1 0x66461f 0x67469e 0x85e7d9 0x662247 0x7fea535aed8e 0x7fea535aee3e 0x669e23 0xfffffffffffffffe
ghdl:error: compilation error
make: *** [../../bin/Makefile:496: testbench] Error 1

```

diogratia, 2022-10-22:
The `make ghdl` target (analyze and elaborate) also fails executing `analyze ./../../lib/gaisler/noelv/core/iunv.vhd` which says analysis fails. 

From the last warning before the crash we can tell the error is after line 10960 (out of 15164 lines):

```none
./../../lib/gaisler/noelv/core/iunv.vhd:10960:26:warning: declaration of "dbgi" hides port "dbgi" [-Whide]
                         dbgi           : in  nv_debug_in_type;
                         ^
```

This is in iunv.vhd procedure debug_module beginning at line 10954, the last subprogram body in the architecture declarative part before the reserved word begin introducing the architecture statement part (architecture rtl of iunv is, line 180 and noting there are no processes containing subprogram bodies. There's a high likelihood the issue is found in procedure debug_module depending on what the procedure Allocate_Complex_Object does when reflected in the assertion error.

Line 302 that cause the assertion in trans-chap4.adb:
```ada
        -- Cannot allocate unconstrained object (since size is unknown).
        Pragma Assert (Type_Info.Type_Mode not in Type_Mode_Unbounded);
```
in procedure Allocate_Complex_Object.

From that you could infer that the type of the object being allocated is unconstrained.

It's possible to fault isolate without a debugger.

By commenting out the entire contents of the architecture statement part we can determine the problem is somewhere in it's concurrent statements.

Un-commenting through the comb process (line 14815) shows the problem resides below.

Un-commenting the generate statement syncrregs shows its not in there.

It's not in the  asyncrregs generate statement either (that gets us to line 15054).

The problem is somewhere in the dis generate statement. We uncomment it, determine the problem still exists then taking a shot in the dark and commenting out the iw_gen generate statement shows the problem is found within it (lines 15126 through 15160).

You could note the multiple instanced entity grlib.cpu_disas has nothing but in mode ports making it difficult to determine which port is causing a problem lacking default values when unconnected.

Adding a component declaration for cpu_disas would allow manipulating the association list gracefully to determine which association element is failing. We'd then examine the formal's type declaration and examine the actual expression and declarations for any named primaries found in the actual. The goal is to determine whether there is a ghdl issue or something found in the downloaded archive. 

The make process appears to fail further on which may be the result of platform (MacOS 12.6) or build environment, something to do with RAM contents.

diogratia, 2022-10-26:
Here's a more minimal reproducer:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity cpu_disas is
generic (
  disasg : in integer range 0 to 3 := 1
  );
port (
  pc            : in  std_logic_vector -- ;                 -- PC
  );
end entity;

architecture behav of cpu_disas is
begin
end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity test is
    generic (
       physaddr     : integer range 32 to 56 := 36; -- Physical Addressing GUESS
        single_issue : integer range 0  to 1 := 1; -- CHANGED set here
        disas        : integer := 1 -- follows entity
    );

end entity;

architecture foo of test is
    function maximum(x : integer; y: integer) return integer is
    begin
      if x > y then
        return x;
      else
        return y;
      end if;
    end;

    function minimum(x : integer; y: integer) return integer is
    begin
      if x < y then
        return x;
      else
        return y;
      end if;
    end;
    -- use gaisler.noelv.XLEN;
    constant XLEN:  integer := 32;  -- tempted to try 42
    constant va:        std_logic_vector := x"abc";
    -- constant va         : std_logic_vector := gaisler.mmucacheconfig.va(riscv_mmu);
    constant pa:        std_logic_vector := x"dead";
    -- constant pa         : std_logic_vector := gaisler.mmucacheconfig.pa(riscv_mmu);

    function addr_bits return integer is
    begin
      return minimum(XLEN, 1 + maximum(va'length, minimum(physaddr, pa'length)));
    end;
    subtype addr_type is std_logic_vector(addr_bits - 1 downto 0);
    subtype pctype         is addr_type;
    constant lanes       : std_logic_vector(0 to 1 - single_issue) := (others => '0');  -- Used as range.
    type pc_lanes_type     is array (lanes'range) of pctype;
    subtype wordx  is std_logic_vector(XLEN - 1 downto 0);
    -- Return high bit from data.
    function get_hi(data : std_logic_vector) return std_logic is
    begin
      return data(data'high);
    end;
    function pc2xlen(pc : pctype) return wordx is
      -- Non-constant
      variable data : wordx;
    begin
      data           := (others => get_hi(pc));
      data(pc'range) := pc;

      return data;
    end;
    
    signal pc             : pc_lanes_type; -- initialize?
begin
    iw_gen: for i in lanes'range generate
      iw : entity work.cpu_disas  -- grlib.cpu_disas
        generic map (
          disasg => disas
          )
        port map(
          pc          => pc2xlen(pc(i)) -- ,
          );
    end generate;
  -- end generate;
end architecture;
```
which produces

```none
%: ghdl -a test.vhdl

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r811.g2ae368913) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Users/diogratia/Desktop/Issue_2228/
Command line:
/Users/diogratia/opt/share/ghdl/bin/ghdl1-llvm -P/Users/diogratia/opt/share/ghdl/lib/ghdl/ieee/v93/ -P/Users/diogratia/opt/share/ghdl/lib/ghdl/ -c -o test.o test.vhdl
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap4.adb:302
Load address: 0x104a1b000
Call stack traceback locations:
0x104d53b91 0x104cfc32e 0x104d0ad37 0x104d0b491 0x104c96ac9 0x104c96dce 0x104c97f6b 0x104c98ae0 0x104d33887 0x104d2f1bc 0x104d2fb1b 0x104d2f719 0x104d2fb9a 0x104ce9d85 0x104cdc281 0x104d3aede 0x104d3c4e5 0x104d3d7c2 0x104a1fd8a
******************************************************************
ghdl:error: compilation error
%:
```

While preserving how the actual is produced.

And in -2008:

```none
%: ghdl -a --std=08 test.vhdl

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r811.g2ae368913) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Users/diogratia/Desktop/Issue_2228/
Command line:
/Users/diogratia/opt/share/ghdl/bin/ghdl1-llvm --std=08 -P/Users/diogratia/opt/share/ghdl/lib/ghdl/ieee/v08/ -P/Users/diogratia/opt/share/ghdl/lib/ghdl/ -c -o test.o test.vhdl
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap4.adb:302
Load address: 0x100624000
Call stack traceback locations:
0x10095cb91 0x10090532e 0x100913d37 0x100914491 0x10089fac9 0x10089fdce 0x1008a0f6b 0x1008a1ae0 0x10093c887 0x1009381bc 0x100938b1b 0x100938719 0x100938b9a 0x1008f2d85 0x1008e5281 0x100943ede 0x1009454e5 0x1009467c2 0x100628d8a
******************************************************************
ghdl:error: compilation error
%:
```
producing the same error, which tells us a bit further analysis is called for.

MarteGit, 2022-10-28:
Thanks a lot for your great attention regarding this issue.

tgingold, 2022-10-29:
I can reproduce the crash and I started to investigate.

dalex78, 2026-09-04:
# Issue #2228 — an `in` conversion that changes the length, on an unconstrained formal port

Analysis by an AI agent. Recommend merging the fix and closing.

## The reproducer

The report pointed at a 15 000-line file inside a 200 MB grlib tarball. diogratia reduced it on the thread. That reduction is kept as `testsuite/gna/issue2228/repro.vhdl`; here is the whole bug in 33 lines:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity sub is
  port (pc : in std_logic_vector);          -- unconstrained formal
end entity;

architecture behav of sub is
begin
  process (pc) begin
    report "sub pc'length = " & integer'image(pc'length);
  end process;
end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity top1 is
end entity;

architecture foo of top1 is
  subtype small is std_logic_vector(16 downto 0);   -- 17 bits
  subtype wordx is std_logic_vector(31 downto 0);   -- 32 bits
  function widen (p : small) return wordx is
    variable data : wordx := (others => '0');
  begin
    data(p'range) := p;
    return data;
  end;
  signal pc : small := (others => '0');
begin
  u : entity work.sub port map (pc => widen(pc));   -- conversion function
end architecture;
```

`pc => widen(pc)` is not an expression: it is VHDL's *conversion function* on the actual part. The formal is unconstrained, and the conversion changes the length -- 17 in, 32 out. That is the whole trigger; the generate statement, the generics and the `pc2xlen`/`get_hi` functions in the original reduction are incidental.

## What GHDL did

Two different wrong answers, one per elaboration path:

| backend | result on master |
|---|---|
| mcode, llvm-jit | `error: mismatching vector length; got 32, expect 17` |
| gcc, llvm | `error: NULL access dereferenced`, no source location |

Neither is right: the design is legal and `sub` should see a 32-bit `pc`.

The llvm/gcc case is memory corruption, not a diagnostic. Under gdb the segfault is in `Grt.Signals.Order_Signal` walking a source list through a garbage pointer (`sig=0x21`), reached from `Order_All_Signals` during `__ghdl_simulation_init`. In the version reported (3.0.0-dev) the same design hit `pragma Assert (Type_Info.Type_Mode not in Type_Mode_Unbounded)` at `trans-chap4.adb:302` instead; the assertion has since moved but the underlying mistake did not.

## What the rule is

GHDL already knows it, and enforces it in `sem`. `vhdl-sem_assocs.adb` quotes it before rejecting a conversion whose result is unconstrained:

>  LRM08 5.3.2.2 Index constraints and discrete ranges
>  e) 3) -- For an interface object or subelement whose mode is IN, INOUT or LINKAGE, if the actual part includes a conversion function or a type conversion, then the result type of that function or the type mark of the type conversion shall define a constraint for the index range corresponding to the index range of the objet, [...]

and the message you get for a conversion function returning an unconstrained type is

```
error: type of actual conversion must be fully constrained
```

So the formal's index range comes from the **result subtype of the conversion**, never from the actual. Both elaborators were reading it from the actual.

## What the fix does

Two one-branch changes, one per path.

1. `src/synth/elab-vhdl_insts.adb`, `Elab_Port_Association_Type` -- the unbounded formal took its bounds from `Exec_Name_Subtype (Get_Actual (Assoc))`, i.e. the actual *name*, with the conversion ignored. It now takes them from the conversion's result subtype when there is an actual conversion. This is the mcode / llvm-jit path, and it is what produced the bogus `expect 17`.

2. `src/vhdl/translate/trans-chap4.adb`, `Elab_Conversion` -- for an unbounded output the conversion's result signal was given bounds by `Chap7.Translate_Type_Conversion_Array_Bounds (…, Out_Type, In_Type, Conv)`, which converts the bounds of the *input*. That is correct for a type conversion, where the length is preserved, and wrong for a function call, where it is not. It now takes the bounds from the result subtype when the conversion is a function call with a fully constrained result.

The second one is what caused the corruption: `Chap5.Get_Unconstrained_Port_Bounds` already gave the formal port the right bounds (32) from the same result subtype, while the conversion's own output signal was built with the actual's (17). The connection then walked 32 signals over a 17-element array. Both ends now agree.

## Testing

`testsuite/gna/issue2228` has both:

- `repro.vhdl` -- diogratia's reduction, analysed, elaborated and simulated;
- `conv.vhdl` -- the same shape, but asserting what the formal actually receives: `pc'length = 32`, the low 17 bits equal to the actual and the padding zero, so a fix that got the length right and the data wrong would not pass.

On master the test fails on every backend -- with `mismatching vector length` on mcode and llvm-jit, and with `NULL access dereferenced` on llvm. With the fix, `sanity`, `gna` and `synth` all pass on mcode, llvm and llvm-jit.

A conversion that does *not* change the length (17 in, 17 out) worked before and still works; that case is what made the bug look erratic in the first place.