ghdl/ghdl #2229: Support for -- RTL_SYNTHESIS ON/OFF

bjourne, 2022-10-23:
It would be great if ghdl could support the LRM-mandated meta comments for turning synthesis on and off. See https://ieeexplore.ieee.org/document/1342563

Paebbels, 2022-10-23:
@bjourne to be correct here, the IEEE Std. 1076.6-2004 `IEEE Standard for VHDL Register Transfer Level (RTL) Synthesis` was withdrawn in 2010 and is no longer active.

The correct name should be:
```vhdl
-- synopsys translate_on/off
-- synthesis translate_on/off
-- pragma translate_on/off
```

I have never seen a `rtl_synthesis on/off`, but you're rights, that what they have written in that document.

bjourne, 2022-10-24:
That's confusing  and I didn't know that. But maybe ghdl could support `RTL_SYNTHESIS ON/OFF` anyway for poor sobs like me who thought it was the correct way. :)

Paebbels, 2022-10-24:
I think, when GHDL supports such a feature in general - which is a need for many designs - it's not so complicated to support all the variants :).