ghdl/ghdl #2234: Internal error when outputting verilog

shenki, 2022-10-28:
**Description**
Internal error when running synth with verilog output, when building the neorv32 soft core.

**Expected behaviour**
Synth should succeed.

**How to reproduce?**

I discovered this when building neorv32 using litex.

I have attached a zipfile containing the vhdl source and a makefile to reproduce.

[neorv32.zip](https://github.com/ghdl/ghdl/files/9884566/neorv32.zip)
```
unzip neorv32
cd neorv32
make
```
I made some attempts to narrow down the issue to a single source file but was unsuccessful. 

**Context**

- OS:
- Origin:
  - [ ] Built from sources: `commit 5cbfa82ebe90ac1016899c9cbecbaf07daa7e588`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r859.g5cbfa82ebe90) [Dunoon edition]
Compiled with GNAT Version: 10.4.0
Target: x86_64-linux-gnu
/home/joel/tmp/neorv32/
Command line:
ghdl --synth --out=verilog --std=08 --no-formal --work=neorv32 -gCONFIG=2 -gDEBUG=False neorv32_bus_keeper.vhd neorv32_busswitch.vhd neorv32_cpu_alu.vhd neorv32_cpu_bus.vhd neorv32_cpu_control.vhd neorv32_cpu_cp_bitmanip.vhd neorv32_cpu_cp_cfu.vhd neorv32_cpu_cp_fpu.vhd neorv32_cpu_cp_muldiv.vhd neorv32_cpu_cp_shifter.vhd neorv32_cpu_decompressor.vhd neorv32_cpu_regfile.vhd neorv32_cpu.vhd neorv32_debug_dm.vhd neorv32_debug_dtm.vhd neorv32_dmem.default.vhd neorv32_fifo.vhd neorv32_icache.vhd neorv32_imem.default.vhd neorv32_litex_core_complex.vhd neorv32_mtime.vhd neorv32_package.vhd neorv32_sysinfo.vhd neorv32_top.vhd neorv32_wishbone.vhd -e neorv32_litex_core_complex
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : netlists-disp_verilog.adb:393
Call stack traceback locations:
0x562ba589e759 0x562ba589f22b 0x562ba589ea6a 0x562ba5a1db3f 0x562ba594b4d6 0x562ba5a2438e 0x562ba585814b 0x7f295e029208 0x7f295e0292ba 0x562ba5856d1f 0xfffffffffffffffe
******************************************************************
```

stnolting, 2022-10-28:
I am not sure at all, but could this be a problem with `ghdl --synth` instead of `ghdl synth`?

I have tested the NEORV32 + GHDL synth feature and it works out of the box: [neorv32-verilog](https://github.com/stnolting/neorv32-verilog)

This is the script I am using: [convert.sh](https://github.com/stnolting/neorv32-verilog/blob/main/src/convert.sh)

shenki, 2022-10-28:
> I am not sure at all, but could this be a problem with `ghdl --synth` instead of `ghdl synth`?

I see the internal error with both forms of the command.

tgingold, 2022-10-29:
Thank you for the testcase, should now be fixed.

commit 36b19897 (2022-10-29): synth: fix crash in disp_verilog.  Fix #2234
closed 2022-10-29 by tgingold