ghdl/ghdl #2235: External signal alias as procedure parameter causes bug in trans-chap6.adb

avelure, 2022-10-30:
**Description**
Using an external signal alias as a input parameter to a procedure causes this bug. I did not see other bugs open for this part of trans-chap6.

**Expected behaviour**
Probably a "cannot handle IIR_KIND_EXTERNAL_SIGNAL_NAME" bug instead.

**How to reproduce?**
```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity;

architecture rtl of test is
  signal sig  : std_logic;
begin
end architecture;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test1 is
end entity;

architecture rtl of test1 is
begin
  i_test : entity work.test;

  p_proc : process
    alias sig    is <<signal g_test.i_test.sig : std_logic>>;
    procedure check(
      signal   clock      : std_logic
    ) is
    begin
    end procedure;
  begin
    check(sig);
    wait;
  end process;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 ent.vhd
ghdl --elab-run --std=08 test
```

**Context**
- OS: Win10
- Origin: Nightly llvm 1cc85c578

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r870.g1cc85c578) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-w64-mingw32
C:\proj\public\minimal\
Command line:
C:\prog\GHDL\nightly-mingw64-llvm\bin\ghdl1-llvm.exe --std=08 -PC:\prog\GHDL\nightly-mingw64-llvm\lib/ghdl\ieee\v08\ -PC:\prog\GHDL\nightly-mingw64-llvm\lib/ghdl\ -c -o test_ext_sig_alisa_proc.o test_ext_sig_alisa_proc.vhd
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : trans-chap6.adb:1083
Load address: 0x7ff60fc60000
Call stack traceback locations:
0x7ff6100433f6 0x7ff60fed2498 0x7ff60fed3f0f 0x7ff60fecc533 0x7ff60fecc510 0x7ff60fed3ff4 0x7ff60ff557aa 0x7ff60ff526f8 0x7ff60ff57949 0x7ff60ff5e7b7 0x7ff60ff5e81b 0x7ff60ff62522 0x7ff60ff688da 0x7ff60ff1ea16 0x7ff60ff10b36 0x7ff60ff74f9c 0x7ff60ff769cc 0x7ff60ff78113 0x7ff60fc62d06 0x7ff60fc613ac 0x7ff60fc614e4 0x7ffbde8e7032 0x7ffbdfd8269f
******************************************************************
```

tgingold, 2022-10-31:
Yes, it is because external names are not yet supported.

donnie-j, 2022-11-07:
Just adding my vote, fwiw, to bump priority a little here.  I'd like to use this feature for debugging on actual hardware, now that synthesis support is quite excellent.  Once this is possible, I'll probably make a quick and dirty on chip/FPGA 'logic analyzer' package one can include in other projects using external names to easily probe around a design.

tgingold, 2022-11-08:
So, do you also need external names for synthesis ?

donnie-j, 2022-11-08:
Yes, I'd like to use external signals for synthesis.  I suppose the question would be how to do that without flattening the hierarchy.  Does yosys have way to access signals internal to a module instantiated in the hierarchy?

tgingold, 2022-11-08:
No, so the design has to be flattened from the start.  This has to be implemented.

commit 147512be (2023-09-18): synth-vhdl_decls: avoid a crash on invalid external names

Fix #2235
closed 2023-09-18 by tgingold