ghdl/ghdl #2236: In synch, evaluate generics early to avoid unwanted dependencies

ozbenh, 2022-10-31:
Hi !

While working on Microwatt, we have toplevel generics that control what IOs to include in the design. Some involve complex modules and dependencies we don't necessarily want to include.

However, ghdl seems to evaluate generics late (in fact in non-synth mode at runtime) such that all the possible modules must be present in the work library regardless of the generics used.

Example, this `foo.vhdl`:
```
library work;

entity toplevel is
    generic (
        HAS_BAR : boolean := false
        );
end entity toplevel;

architecture rtl of toplevel is
begin
    maybe_bar : if HAS_BAR generate
    begin
        bar: entity work.bar;
    end generate;
end architecture;
```
Results in:
```
$ ghdl --synth --out=verilog -gHAS_BAR=false foo.vhdl -e toplevel 
foo.vhdl:13:26:error: unit "bar" not found in library "work"
        bar: entity work.bar;
                         ^
```
Ideally we would want to be able to include build in the library the modules required for the specific synth target as established by the generics. This can be important when such modules might be themselves complex things, possibly imported from elsewhere or not even available as vhdl sources (for example when using verilator and mixing languages).

tgingold, 2022-10-31:
Well, no.  This won't be compliant with the LRM.
The right way to do this is either to use components, or different architectures.

ozbenh, 2022-10-31:
Hrm... so the languate requires that everything inside of a `if .... generate` statement be able to build and has all of its dependencies even if the condition of the `if` is false ? That's ... annoying ;)

In my specific case, microwatt targetted at verilator, with or without the litesdram simulation. The top-level need to include the l2 "wrapper" vhdl (which itsels uses litedram .v as a component) when using litedram, or not include it ... when not. This doesn't seem possible from what you say. We use that consruct all the time with Xilinx :-) but ok...

What would be the best alternative ? I could have a separate l2 wrapper implementation that's basically empty and build that into work instead when generating that sim model ?

tgingold, 2022-10-31:
Use a component and instantiate the component instead of the entity.  That's the simplest solution.

closed 2022-11-05 by tgingold