ghdl/ghdl #2238: Type qualifier + a literal in a port map erroneously causes a compile error (VHDL-93)

JimLewis, 2022-11-02:
**Description**
For the following code results in the GHDL error, "actual expression must be globally static".
Note if the type qualifier is removed, then the code works fine, however, it is being 
generated by an automatic generator.   

```vhd :file: PortMap2.vhd
library IEEE ;
use ieee.std_logic_1164.all ; 
use ieee.numeric_std.all ; 

entity PortMap2 is
end entity PortMap2 ;
architecture t1 of PortMap2 is 
  component fred is
    port (A : in signed(7 downto 0) ; B : out signed (7 downto 0) ) ; 
  end component fred ; 
  
  signal Result : signed (7 downto 0) ; 
begin

  fred1 :  fred 
    port map (A => signed'("00001111"),  B => Result ) ; 

end architecture t1 ; ```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a PortMap2.vhd
```
**LRM References**
1076-1993:
**Description**
a) A literal of any type other than type TIME
h) A qualified expression whose operand is a locally static expression

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

closed 2022-11-02 by tgingold

JimLewis, 2022-11-02:
@tgingold  That one was only open 8 hours.   👍

Dolu1990, 2022-11-02:
Thanks for the fix :D

commit bf5c6cda (2022-11-02): vhdl-sem_expr(sem_qualified_expression): relax staticness rules.

Fix #2238