ghdl/ghdl #2253: Bound check failure for unconstrained records in port

avelure, 2022-11-19:
**Description**
There is an issue with determining bounds on unconstrained vectors in a record in ports.

**Expected behaviour**
I expect the code to simulate without a bounds error.

**How to reproduce?**
An MWE is below.

One of the vectors in the MWE is a null array, but the issue still happens if I change the null array to non-null.

In the original code, the simulation would end with just a `.` on the last line and the return code being `-1` without displaying a bounds error.  If I deletes some arbitrary lines from the code the bounds would consistently appear, but there didn't seem to be a rhyme or reason to which lines needed to be deleted, so I guess there is also some stack or memory issue. I have a much bigger MWE that recreates that, which I can post if the fix to this issue does not solve the original problem.

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test_pkg is
  type t_axi_wr_slave_in_if is record
    awid      : std_logic_vector;
    awaddr    : std_logic_vector;
    awlen     : std_logic_vector(7 downto 0);
    awsize    : std_logic_vector(2 downto 0);
    awburst   : std_logic_vector(1 downto 0);
    awlock    : std_logic;
    awcache   : std_logic_vector(3 downto 0);
    awprot    : std_logic_vector(2 downto 0);
    awqos     : std_logic_vector(3 downto 0);
    awregion  : std_logic_vector(3 downto 0);
    awuser    : std_logic_vector;
    awvalid   : std_logic;
    wdata   : std_logic_vector;
    wstrb   : std_logic_vector;
    wlast   : std_logic;
    wuser   : std_logic_vector;
    wvalid  : std_logic;
    bready  : std_logic;
  end record;
  type t_axi_wr_slave_out_if is record
    awready : std_logic;
    wready  : std_logic;
    bid     : std_logic_vector;
    bresp   : std_logic_vector(1 downto 0);
    buser   : std_logic_vector;
    bvalid  : std_logic;
  end record;
  type t_axi_rd_slave_in_if is record
    arid      : std_logic_vector;
    araddr    : std_logic_vector;
    arlen     : std_logic_vector(7 downto 0);
    arsize    : std_logic_vector(2 downto 0);
    arburst   : std_logic_vector(1 downto 0);
    arlock    : std_logic;
    arcache   : std_logic_vector(3 downto 0);
    arprot    : std_logic_vector(2 downto 0);
    arqos     : std_logic_vector(3 downto 0);
    arregion  : std_logic_vector(3 downto 0);
    aruser    : std_logic_vector;
    arvalid   : std_logic;
    rready  : std_logic;
  end record;
  type t_axi_rd_slave_out_if is record
    arready : std_logic;
    rid     : std_logic_vector;
    rdata   : std_logic_vector;
    rresp   : std_logic_vector(1 downto 0);
    rlast   : std_logic;
    ruser   : std_logic_vector;
    rvalid  : std_logic;
[... 212 lines cut ...]
      GC_ADDR_WIDTH_1 => C_ADDR_WIDTH_1,
      GC_DATA_WIDTH_1 => C_DATA_WIDTH_1,
      GC_ID_WIDTH_1   => C_ID_WIDTH_1,
      GC_USER_WIDTH_1 => C_USER_WIDTH_1
    );
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-llvm
ghdl -a --std=08 ent.vhd
ghdl --elab-run --std=08  test
```

```
ghdl --elab-run --std=08 test
.\test:error: bound check failure at test_port_map_interface_5.vhd:147
in process .test(beh).i_test3@test3(beh).i_test@test2(beh).P0
.\test:error: simulation failed
```

**Context**
Please, provide the following information:

- OS: Win10
- Origin: nightly llvm 1cc85c578

Paebbels, 2024-12-19:
@avelure do you mind rechecking with latest GHDL?

avelure, 2024-12-20:
Sure, works fine now.

closed 2024-12-20 by Paebbels