ghdl/ghdl #2254: Complex generic in when statement - ```choice is not locally static```

thmslg, 2022-11-25:
**Description**
GHDL flags GENERIC as not locally static.

**Expected behaviour**
Expressions should be considered to be locally static as long as they can be resolved to a constant at compile time

**How to reproduce?**

```library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test_ghdl is
  generic (
    N        : positive := 32
    );
  port (
    clk_i   : in  std_logic := 'X';
    dummy_o : out std_logic
  );
end test_ghdl;

architecture rtl of test_ghdl is
  signal state_reg  : natural range N downto 0 := 0;
begin
  process (clk_i) is
  begin
    if clk_i'event and clk_i = '1' then
      case state_reg is
        when (N) => --Error is coming from here error: choice is not locally static
          dummy_o    <= '1';
          state_reg <= state_reg - 1;
        when others =>
          state_reg <= 0;
          dummy_o    <= '0';
      end case;
    end if;
  end process;
end architecture rtl;
```

```ghdl -a --std=08 test_ghdl.vhd
tb.vhd:22:9:error: choice is not locally static
        when (N) => --Error is coming from here error: choice is not locally static
        ^
```

**Context**
Please, provide the following information:

- OS: Ubuntu 20.04
- Origin: Built from sources: `93a88845da27290669c0dee990b7bb7232d58139`

**Additional context**
Seems a really similar issue as [1486](https://github.com/ghdl/ghdl/issues/1486)

Thanks for your work !
BR,
T.

diogratia, 2022-11-25:
See IEEE Std 1076-2008 9.4 Static expressions

>There are two categories of static expression. Certain forms of expression can be evaluated during the analysis of the design unit in which they appear; such an expression is said to be *locally static*. Certain forms of expression can be evaluated as soon as the design hierarchy in which they appear is elaborated; such an expression is said to be *globally static*.

The hazards of carrying precepts in from other programming languages:

"Expressions should be considered to be locally static as long as they can be resolved to a constant at compile time"
 
Here 'compile' encompasses both analysis and elaboration.  Your error message shows up during analysis, N is not locally static.

N is globally static. After reviewing 9.4.2 Locally static primaries see 9.4.2 Globally static expressions "c) A generic constant declared with a globally static subtype". Other useful subsections include 6.5.2 Interface object declarations the definition of *interface object* and *interface constant*, 6.5.6.2 Generic clauses.
 
The rule is given in 10.9 Case statement "All simple expressions and discrete ranges given as choices in a case statement shall be locally static."

The reason for requiring locally static choices is to insure each value of the case expression is represented once and only once in the set of choices at analysis time (10.9):

>For an ordinary case statement, or for a matching case statement in which the expression is of type BIT or an array type whose element type is BIT, if the expression is the name of an object whose subtype is locally static, whether a scalar type or an array type, then each value of the subtype shall be represented once and only once in the set of choices of the case statement, and no other value is allowed; this rule is likewise applied if the expression is a qualified expression or type conversion whose type mark denotes a locally static subtype, or if the expression is a call to a function whose return type mark denotes a locally static subtype, or if the expression is an expression described in this paragraph and enclosed in parentheses.

tgingold, 2022-11-30:
Yes, I agree.  The reproducer is not valid.

closed 2022-11-30 by tgingold