ghdl/ghdl #2258: Option to send warnings to stdout

avelure, 2022-12-01:
**Is your feature request related to a problem? Please describe.**
I have some tooling that that looks for errors in the stdout/stderr and suppresses the rest during compile and sim. I see that warnings get printed on stderr, it would be useful if there was a commandline option to send them to stdout instead.

**Describe the solution you'd like**
Either an option to only redirect warning to stdout or an option to choose from which error level the redirection to stderr starts, kind of like `--assert-level`.

**Describe alternatives you've considered**
I could do some regex to filter it ofcourse, but it is complicated by the fact that each message is several lines.

tgingold, 2022-12-01:
For analysis: you can disable all the warnings, and therefore you will get only error messages.
For simulation: you can also disable asserts.

Does that help ?

avelure, 2022-12-01:
For analysis: It is an option, but as far as I can see there is no global switch so I need to disable each one and assume there will be no new warnings implemented in the future. Additionally, I still might like to log the warnings to screen/file without acting upon them as errors.
For simulation: Wouldn't this also disable asserts for error/fatal?

tgingold, 2022-12-01:
There is -Wall and -Wno-all to enable/disable all the warnings.

For assertions, maybe we can add a new flag to silent them until a level.

avelure, 2022-12-01:
> There is -Wall and -Wno-all to enable/disable all the warnings. 
This is an option, though it seems to be missing from the documentation.

> For assertions, maybe we can add a new flag to silent them until a level.
I'm not so much interested in silencing them completely, as I do some automatic reporting on the number of warnings and errors at the end, comparing them to the expected.

umarcor, 2023-03-08:
FTR: https://github.com/ghdl/ghdl/commit/9f59cc8640127459cc09f3110f0dcbad4000ed2d