ghdl/ghdl #2264: Type mismatch with generic package instance

creiter64, 2022-12-06:
**Description**
In the following example I try to pass a generic package instance as a generic parameter to an entity.
Unfortunately ghdl aborts with the following error message:
```
testbench/test_pattern_generator_tb.vhd:87:29:error: can't associate 'control_sig' with port "options"
            Options      => control_sig
                            ^
testbench/test_pattern_generator_tb.vhd:87:29:error: (type of 'control_sig' is test_pattern_options_t)
testbench/test_pattern_generator_tb.vhd:39:9:error: (type of port "options" is test_pattern_options_t)
```
Which I guess means it detects the types as different.

**Expected behaviour**
I expect this example to compile fine, since the types are the same AFAIK. Additionally Questa seems to have no issue with this construct.

**How to reproduce?**
Minimum example to reproduce:

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

--! Configuration data types and structures
package test_pattern_generator_generic_pkg is
    generic(
        PIXEL_PER_LINE_MAX : positive := 640
    );

    --! Redeclared constants so that we don't have to pass them to dependent entities as generics again
    constant C_PIXEL_PER_LINE_MAX : positive         := PIXEL_PER_LINE_MAX;

    --! Constrained types with valid ranges for run time configurable options
    subtype Pixel_Per_Line_t is integer range 2 to PIXEL_PER_LINE_MAX;

    type Test_Pattern_Options_t is record
        PixelPerLine  : Pixel_Per_Line_t;
    end record Test_Pattern_Options_t;

    --! Default options for a test pattern generator that are guaranteed to produce valid output
    constant DEFAULT_OPTIONS : Test_Pattern_Options_t := (
        PixelPerLine  => Pixel_Per_Line_t'low
    );

end package test_pattern_generator_generic_pkg;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test_pattern_generator is
    generic(
        package GENERIC_PKG_INSTANCE is new work.test_pattern_generator_generic_pkg generic map(<>)
    );
    port(
        Clk          : in  std_logic;   -- Clock Signal
        --! Generated image parameters runtime options
        Options      : in  GENERIC_PKG_INSTANCE.Test_Pattern_Options_t
    );

end test_pattern_generator;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of test_pattern_generator is
    use GENERIC_PKG_INSTANCE.all;
    --! Pixel counter, 0 based up-counter, counts pixels in the current line
    signal pixel_cnt : natural range 0 to GENERIC_PKG_INSTANCE.C_PIXEL_PER_LINE_MAX - 1;

begin
end architecture rtl;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test_pattern_generator_tb is
end entity test_pattern_generator_tb;

architecture sim of test_pattern_generator_tb is
    constant CLK_PERIOD : time := 10 ns;

    -- Test Pattern Generator Generics
    constant PIXEL_PER_LINE_MAX : integer  := 160;

    -- Instance for the type package for the test pattern generator
    package test_pattern_pkg_inst is new work.test_pattern_generator_generic_pkg generic map(
            PIXEL_PER_LINE_MAX => PIXEL_PER_LINE_MAX
        );
    use test_pattern_pkg_inst.all;

    -- Test Pattern Generator Inputs
    signal clk          : std_logic                                    := '0';
    signal control_sig  : test_pattern_pkg_inst.Test_Pattern_Options_t := test_pattern_pkg_inst.DEFAULT_OPTIONS;
    
begin
    test_pattern_generator_inst_tb : entity work.test_pattern_generator
        generic map(
            GENERIC_PKG_INSTANCE => test_pattern_pkg_inst
        )
        port map(
            Clk          => clk,
            Options      => control_sig
        );

    clk <= not clk after CLK_PERIOD / 2;

    test_runner : process
    begin
        wait for CLK_PERIOD;
    end process test_runner;

end architecture sim;

```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08  testbench/test_pattern_generator_tb.vhd
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS: Happens on Linux/Windows
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `GHDL 3.0.0-dev (2.0.0.r948.g93cb47c44) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 mcode code generator
Written by Tristan Gingold.`

**Additional context**
I'm not an VHDL expert, so I don't know if this construct is even legal my only reference is that it works in Questa. I'd also be happy about advice on how to workaround this issue.

diogratia, 2022-12-11:
6.2 Type declarations

>The types created by the elaboration of distinct type definitions are distinct types.  ...

4.9 Package instantiation declarations

>A package instantiation declaration defines an instance of an uninstantiated package. The instance is called an *instantiated package*.  
> ...  
>  
>The package instantiation declaration is equivalent to declaration of a generic-mapped package, consisting of a package declaration and possibly a corresponding package body. The simple name of the generic-mapped package declaration is the identifier of the package instantiation declaration. The generic-mapped package declaration has the generic clause of the uninstantiated package declaration, the generic map aspect of the package instantiation declaration, and the declarations of the uninstantiated package declaration. The package body corresponding to the generic-mapped package is present if the uninstantiated package has a package body. In that case, the simple name of the generic-mapped package body is the identifier of the package instantiation declaration, and the declarations of the generic-mapped package body are the declarations of the uninstantiated package body. The meaning of any identifier appearing anywhere in the generic-mapped package declaration or package body is that associated with the corresponding occurrence of the identifier in the package instantiation declaration, the uninstantiated package declaration, or the uninstantiated package body, respectively, except that an identifier that denotes the uninstantiated package denotes, instead, the generic-mapped package.

The two declarations of the instantiated packages types Test_Pattern_Options_t are distinct from each other.

tgingold, 2022-12-11:
Yes, but isn't the package instantiated only once ?

diogratia, 2022-12-11:
Tristan makes a valid point, I ruined the previous analysis by using modified code.

6.5.5 Interface package declarations tells us the interface package declaration  (the formal) can be associated with any instantiation of the uninstantiated package and that it makes the instantiated package visible by the instantiation name. 

That'd make one declaration of type Test_Pattern_Options_t in the instantiated package as an entity test_pattern_generator_tb architecture declarative item the name of the instantiated package test_pattern_pkg_inst.

I also tried to instantiate the package as a primary design unit and using a component declaration for test_pattern_generator and instantiating that. Both got the same error messages.

Commented out the port options and the association with it as well as the declaration for control_sig to see if the interface generic package declaration would work without running into the error. Got a crash:

```none
%: ghdl -a --std=08 test_pattern_generator_tb.vhdl

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r811.g2ae368913) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Users/dio_gratia/Desktop/
Command line:
/Users/dio_gratia/opt/share/ghdl/bin/ghdl1-llvm --std=08 -P/Users/dio_gratia/opt/share/ghdl/lib/ghdl/ieee/v08/ -P/Users/dio_gratia/opt/share/ghdl/lib/ghdl/ -c -o test_pattern_generator_tb.o test_pattern_generator_tb.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:553
Load address: 0x10203a000
Call stack traceback locations:
0x102113552 0x10211386c 0x1021139e3 0x1022b78b5 0x1022b7acc 0x102352887 0x10234e1bc 0x10234eb1b 0x102308d85 0x1022fb281 0x102359ede 0x10235b4e5 0x10235c7c2 0x10203ed8a
******************************************************************
ghdl:error: compilation error
```

That's in procedure Find_Scope in package Chap10 (from -1993, Scope and visibility).

And that might imply the generic interface package declaration actually got or was assumed to be instantiated instead of the interface generic package name (the formal) denoting the generic mapped actual.
 
I've run out of ideas on getting more information without using a debugger.

creiter64, 2022-12-13:
@diogratia If you could post the code that triggered the crash, I think @tgingold might be able to reproduce it with less effort.

tgingold, 2022-12-14:
Not really needed.  Entities with generic types or packages are not yet supported for simulation, but they should be for synthesis.
I hope to add support for them next year.

creiter64, 2022-12-22:
@tgingold Would it be possible for you to more precisely estimate when this will land in master? Because if this is a huge amount of work and will take months to implement, I'll rather rewrite my code to not use this feature.

tgingold, 2022-12-23:
I understand.  Let me try to add support for what you have during a week.

tgingold, 2022-12-23:
Ok, your testcase is now working correctly.

tgingold, 2022-12-23:
Thank you for the issue.

closed 2022-12-23 by tgingold

creiter64, 2023-01-09:
Thank you for fixing this, I tested it with the full project and everything works as expected.

commit 75717123 (2022-12-23): trans-chap5.adb: handle package association in entities.

Fix #2264