ghdl/ghdl #2266: sync_abort yields "cannot handle N_SYNC_ABORT" in synthesis

reprographix, 2022-12-07:
**Description**
I was trying to do a little bit of formal in VHDL with GHDL+SymbiYosys when I realized that sync_abort was preventing synthesis.

**Expected behaviour**
The assertion is properly synthesized.

**How to reproduce?**

```vhd :file: reproducer.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;

entity reproducer is
    generic(
        DUT_IN_DATA_WIDTH : natural
    );
    port(
        clk : in std_logic;
        reset : in std_logic
    );
end entity;

architecture reproducer_arch of reproducer is
    attribute anyconst  : boolean;
    attribute anyseq    : boolean;

    signal w_dut_in_valid   : std_logic;

    attribute anyseq of w_dut_in_valid  : signal is true;
begin

    default clock is rising_edge(clk);
    a_incr_not_eop: assert always {w_dut_in_valid} |=> {w_dut_in_valid} sync_abort (reset = '1');

end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl --synth --std=08 reproducer.vhd -e reproducer
```

**Context**

- OS: Ubuntu 20.04 LTS
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `93cb47c448978def5fc72674fae13a2622190695`

```
build_property_fa: cannot handle N_SYNC_ABORT

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r948.g93cb47c44) [Dunoon edition]
Compiled with GNAT Version: 9.4.0
Target: x86_64-linux-gnu
/home/jthomas/Documents/Notes/formal_demo/
Command line:
ghdl --synth --std=08 reproducer.vhd -e reproducer
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : psl-errors.adb:37
Call stack traceback locations:
0x55f2654b1980 0x55f2654c99ec 0x55f2654c9805 0x55f2654c9931 0x55f2654c9a12 0x55f2656b856a 0x55f2656b86b5 0x55f2656b95d0 0x55f2656b9a22 0x55f2656bc18e 0x55f2656959f8 0x55f2656967e3 0x55f265696a76 0x55f2656bf507 0x55f26571d808 0x55f26571e30a 0x55f265615873 0x55f2657d089d 0x55f26539bded 0x7f09eed6c081 0x55f26539a2dc 0xfffffffffffffffe
******************************************************************
```

tgingold, 2022-12-07:
Now fixed, thank you for the issue.

You could use 'abort' instead as they are synonymous.

closed 2022-12-07 by tgingold

tmeissner, 2022-12-11:
@tgingold AFAIK `async_abort` is synonym to `abort`.

The PSL standard says:

> 6.2.1.5.1 abort, async_abort, and sync_abort
> The abort, async_abort, and sync_abort operators, shown in Syntax 6-28, specify a condition
> that removes any obligation for a given FL Property to hold. The sync_abort operator expects the abort
> condition to occur in a cycle in which the context clock holds. The abort and async_abort operators
> accept asynchronous abort conditions as well.

commit 9ac2a43a (2022-12-07): psl-build.adb: handle sync_abort.  Fix #2266