ghdl/ghdl #2272: Self contained executable for linux

HendrikMennen, 2022-12-17:
Hello,

I'm working on the VHDPlus Project (vhdplus.com) and I would like to make the use of GHDL as easy as possible for all platforms, to make the use of GHDL more beginner friendly. For linux and macos this works fine. Is there a way to distribute a package that runs on most (if not all) linux machines?

umarcor, 2022-12-17:
OCI containers do run on most (if not all) linux machines.
You can try other container solutions, such as AppImage, Flatpak, Snap...
Or you can try building GHDL statically.

tgingold, 2022-12-17:
Indeed, if you build ghdl with an old distribution it should work everywhere.
The only dependency can be glibc.

HendrikMennen, 2022-12-18:
Using the ubuntu 18 executable, users have issues with libgnat-7, which can't be installed from the package manager on newer distributions

tgingold, 2022-12-19:
Which back-end are you compiling for ?

Try to add `-static` to `GNAT_BARGS` variable in Makefile.

HendrikMennen, 2022-12-19:
I was just using the nightly builds distributed here. Will try to make a build myself

HendrikMennen, 2023-01-10:
seems to work with -static , thank you!

closed 2023-01-10 by HendrikMennen