ghdl/ghdl #2273: GHDL crashes when converting verilog file

jotego, 2022-12-17:
**Description**

When converting the files attached in the zip to verilog with:
```
ghdl -a -fsynopsys  @mc8051.f
ghdl synth -fsynopsys --out=verilog mc8051_core > mc8051.v
```
I get a crash dump (see below)

**Expected behaviour**
I would expect ghdl to either convert the design or report an error in a given source file.

**How to reproduce?**
Unzip the attached file and execute

```
ghdl -a -fsynopsys  @mc8051.f
ghdl synth -fsynopsys --out=verilog mc8051_core > mc8051.v
```

**Context**
I am using the docker image ghdl/ghdl **tag** ubuntu22-llvm-11

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (v2.0.0-952-gd908ceeed) [Dunoon edition]
Compiled with GNAT Version: 10.4.0
Target: x86_64-linux-gnu
/mnt/
Command line:
ghdl synth -fsynopsys --out=verilog mc8051_core
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x56310003ce50 0x5631001250e3 0x563100125db8 0x56310013f506 0x563100156b91 0x56310015818a 0x563100158282 0x56310010d96c 0x56310013f1d9 0x56310010e13f 0x56310013de91 0x5631001526e5 0x56310015dce0 0x563100152ccd 0x56310015dc92 0x563100152ccd 0x56310015dc92 0x56310015bd16 0x56310015dd2c 0x563100152d3d 0x56310015dc92 0x563100152ccd 0x56310015dc92 0x56310015bd16 0x56310015dd2c 0x56310015ef4a 0x563100160914 0x563100160b51 0x563100147932 0x563100147b86 0x56310010e62f 0x563100203780 0x5631000e04fc 0x56310020c7a2 0x5630fff5f3d7 0x7fc222c3fd8e 0x7fc222c3fe3e 0x5630fff5df83 0xfffffffffffffffe
******************************************************************
```
[8051-ghdl.zip](https://github.com/ghdl/ghdl/files/10252279/8051-ghdl.zip)

**Additional context**
None

closed 2022-12-18 by tgingold

jotego, 2022-12-18:
Thank you!
I am a bit ashamed to ask but compiling ghdl is a bit complex for me so I cannot test it straight away. Could you paste the output mc8051.v file for me in a comment in this thread, please?

tgingold, 2022-12-18:
This doesn't scale.  You can get nighly prerelease from the code page on github.

jotego, 2022-12-18:
I will wait for it. Thanks

umarcor, 2022-12-18:
José, CI will create builds automatically and upload them to the nightly pre-release. So, approximately 35-40 min after Tristan pushes something to main, you can get updated artifacts from https://github.com/ghdl/ghdl/releases/tag/nightly. In order to know whether it was updated, compare the commit hash shown there with the one in main:

[image]

[image]

The date of the nightly pre-release is not correct. It shows when we created it first time, but it is not updated when the tag is pushed forward.

Other than that, you can use docker containers. Then, "updating" is just a matter of pulling the image again. No need to "reinstall" GHDL.

jotego, 2022-12-18:
Thanks a lot for the detailed explanation :-)

commit 36c27636 (2022-12-18): synth: factorize code (Exec_Name_Subtype).  Fix #2273