ghdl/ghdl #2274: How to simulate a design containing vendor IPs?

efeyitim, 2022-12-19:
Hi everyone,

I am trying to simulate a design containing some Altera IPs but for all of the IPs GHDL give me a warning `"instance "IP" of component "IP" is not bound"`. So, I just created a PLL and tried to do simulations on it. I did successfully run compile-altera.ps1 to compile Altera libraries and I show the resulting files to GHDL while analyzing the PLL files. You can see the resulting error below when I run the simulation script which I shared at the end of this post.

`D:\Projects\ghdl_ip>ghdl -e --work=work --workdir=work -P=C:\ghdl-2.0.0\compiled_vendor_lib\altera\altera_mf\v93 -fsynopsys -fexplicit tb_pll

pll.vhd:26:9:warning: instance "pll_inst" of component "pll_0002" is not bound [-Wbinding]
             pll_inst : component pll_0002
        
pll.vhd:20:14:warning: (in default configuration of pll(rtl))`

There is a Verilog file related with PLL in a subfolder named pll at the folder where I created the PLL IP. I know the issue is related to this verilog file, I should somehow show it to the GHDL but I do not know how.

 You can see the code for PLL, testbench, my simulation script and resulting error below. My setup is Windows 10 with GHDL 3.0.0.

[PLL code](https://gist.github.com/efeyitim/89b561d54d4803cafc72a51a33a50a21)

[PLL Verilog code](https://gist.github.com/efeyitim/043297f6ecbbb6bca69b06a06af6e86b)

[Testbench](https://gist.github.com/efeyitim/ca718c1f6372baac144fdc093291324b)

[Script](https://gist.github.com/efeyitim/75c1c67f7c55d1a9cba8fac3701519bb)

tgingold, 2022-12-19:
I fear you are stuck.

pll_0002 isn't complicated and could be easily translated to vhdl, but what about altera_pll ?  Do you have a model of it ?

efeyitim, 2022-12-19:
Yes, pll_0002 isn't complicated to translate to VHDL. I do not know much about altera_pll and in which altera library it is located but I think it can be in altera_mf library. As I mentioned in the original post, I have compiled the altera libraries using the script `compile-altera.ps1`, but I do not know if it is enough or not. That is why I have written here. Thanks!

efeyitim, 2022-12-19:
> I fear you are stuck. pll_0002 isn't complicated and could be easily translated to vhdl, but what about altera_pll ? Do you have a model of it ?

Yes, pll_0002 isn't complicated to translate to VHDL. I do not know much about altera_pll and in which altera library it is located but I think it can be in altera_mf library. As I mentioned in the original post, I have compiled the altera libraries using the script compile-altera.ps1, but I do not know if it is enough or not. That is why I have written here. Thanks!

tgingold, 2022-12-20:
I have barely used Quartus so I don't know the answer.  You should investigate!

closed 2024-12-19 by Paebbels