ghdl/ghdl #2278: Bound check failure due to memory corruption?

abyszuk, 2022-12-21:
**Description**
I'm seeing a regression in my test suite, which I believe is caused by some strange memory corruption in GHDL.
I've verified my test suite with other simulators (Questa, Riviera) and older version of GHDL (~October 2022) - it works fine there.
The error I get is bound check failure:
```
/afs/cern.ch/work/a/adbyszuk/praca/cce/cce/vunit_out/test_output/uut.tb_wb_pll.wbfreq=18453323_sfreq_=3000000_spfreq=50_cntrw=20.test_old_flags_90cc2e173b96e81a64eacd55cd152fce6aa122cc/ghdl/tb_wb_pll-tb:error: bound check failure at /home/adrian/.local/lib/python3.9/site-packages/vunit/vhdl/verification_components/src/bus_master_pkg-body.vhd:180
in process .tb_wb_pll(tb).main
/afs/cern.ch/work/a/adbyszuk/praca/cce/cce/vunit_out/test_output/uut.tb_wb_pll.wbfreq=18453323_sfreq_=3000000_spfreq=50_cntrw=20.test_old_flags_90cc2e173b96e81a64eacd55cd152fce6aa122cc/ghdl/tb_wb_pll-tb:error: simulation failed
```
... which is impossible. I'll try to navigate you through the codebase to show what I believe is happening.
1. TB calls a `check_new_data_flags` procedure a few times:
https://gitlab.cern.ch/cce/cce/-/blob/master/common/vhd/timing/pll/tb/tb_wb_pll.vhd#L222
This procedure does nothing more than call VUnit `read_bus()` procedure with the same parameters a few times:
https://gitlab.cern.ch/cce/cce/-/blob/master/common/vhd/timing/pll/tb/tb_wb_pll.vhd#L157
2. At some point, GHDL claims there's bound check error in Vunit `read_bus()` procedure:
https://github.com/VUnit/vunit/blob/master/vunit/vhdl/verification_components/src/bus_master_pkg-body.vhd#L178
I've modified this code to print possible suspects:
```
177   ¦ request_msg := new_msg(bus_read_msg);
178   ¦ report "p_address_length = " & integer'image(bus_handle.p_address_length) & CR & LF; 
179   ¦ report "address'length = " & integer'image(address'length) & CR & LF;
180   ¦ full_address(address'length-1 downto 0) := address;
```
Now, if I filter through the prints from this code, this is the output:
```
bus_master_pkg-body.vhd:178:5:@40006968005493fs:(report note): p_address_length = 32
bus_master_pkg-body.vhd:179:5:@40006968005493fs:(report note): address'length = 32

bus_master_pkg-body.vhd:178:5:@40007076387057fs:(report note): p_address_length = 32
bus_master_pkg-body.vhd:179:5:@40007076387057fs:(report note): address'length = 32

(...)

bus_master_pkg-body.vhd:178:5:@40007347340967fs:(report note): p_address_length = 32
bus_master_pkg-body.vhd:179:5:@40007347340967fs:(report note): address'length = 32685  ### <== THIS!
```
So, at some point in time the `address'length` in the `read_bus()` call gets corrupted!
Value reported differs with each run, but it's always above 32000.

If add more dummy `read_bus()` calls in my `check_new_data_flags` procedure I can make this bug disappear or reappear again.

Please let me know if I can somehow help you debug this further.

**How to reproduce?**
I wasn't able to create a reliable MWE for this. The bug is sometimes hard to trigger - changing anything in the VHDL code may make it disappear. But if it does trigger, it does so reliably.

**Context**
- OS: Docker `ghdl/vunit:gcc` image
- Origin: latest version of this image

tgingold, 2023-01-11:
Until now I was not able to reproduce the crash.  Still trying.
You can also use gdb to narrow the issue.

abyszuk, 2023-05-04:
Hi, I wanted to refresh this issue as I can still reproduce it with current GHDL master and GCC backend. Interestingly my testbench also fails with mcode backend, but the error is completely different.
Could this be connected to issue #2417 ?
I've tested this with both self-made GHDL Docker image and `vunit/ghdl` images, so I'm pretty confident this is 

Anyway, steps to reproduce:
1. Compile GHDL with GCC backend
2. `pip3 install vunit-hdl`
3. `git pull ssh://git@gitlab.cern.ch:7999/cce/cce.git`
4. `cd cce`
5. `python3 common/vhd/interfaces/one_wire/tb/run.py uut.tb_bus_one_wire.clk_frequency_g=3000000,use_tmr_bram_g=True,test_too_many_g=False,bus_type_g=wb.ds1822 --log-level debug`

tgingold, 2023-05-09:
I tried to reproduce, but without success.  Should I also add the assertions ?

abyszuk, 2023-05-09:
Assertions don't change the outcome in my case. This is becoming very intriguing.

Did you try it with `vunit/ghdl` Docker image or with your locally built GHDL?
Just to compare notes, can you tell me what GCC and GNAT version do you use to build GHDL?
For comparison:
- my GHDL - GNAT 2017, GCC 10
- `vunit/ghdl` image - GNAT 9, GCC 9.1
I'm not sure how Debian GNAT version numbering translates to upstream GNAT installer.

tgingold, 2023-05-09:
I don't use docker, only locally built ghdl.  I used gcc 9 on centos 7.

abyszuk, 2023-05-10:
And what's the GNAT version?
I did more tests and can reliably trigger this even if GHDL is built with different versions of GCC (9.5. 10.2, 12.1).
I'd say the most problematic thing is that I can also reliably trigger this with `vunit/ghdl` image, which AFAIK is a kind of official one (at least that's the impression).
How do we proceed? I can try poking with GDB, but I'd need some pointers on what to look for.

commit 45088821 (2023-05-10): trans-chap8(Need_Bounds_Field): adjust conditions

A locally static literal may still need produce code.

Fix #2278
closed 2023-05-10 by tgingold