ghdl/ghdl #2288: TYPES.INTERNAL_ERROR when calling a function within a qualified expression's aggregate

maleiter, 2022-12-26:
**Description**
GHDL Bug occurs if a function is called within the aggregate of a qualified expression. 

**How to reproduce?**
The following Code can be used to reproduce the bug. The example contains an alternative code as a comment which can be used to work around the bug.  When the function fun is called directly within the aggregate, a GHDL Bug occurs. If the function's return value is stored in a variable first, everything is fine. 

```vhd :file: tb_minimal.vhd
library ieee;
use ieee.std_logic_1164.all;

entity tb_minimal is
end tb_minimal;

architecture rtl of tb_minimal is
begin
  MAIN_PROC : process

    function fun return std_ulogic_vector is
      variable v_return : std_ulogic_vector(0 to 3) := x"A";
    begin
      return v_return;
    end function fun;

    variable v_result       : std_ulogic_vector(0 to 7);
    variable v_some_data    : std_ulogic_vector(0 to 3) := x"B";
    variable v_fun_result : std_ulogic_vector(0 to 3);
  begin

    -- working:
    -- v_fun_result := fun;
    -- v_result     := std_ulogic_vector'(v_some_data, v_fun_result);

    -- not working:
    v_result       := std_ulogic_vector'(v_some_data, fun);

    report "result= 0x" & to_hstring(v_result);
    wait;
  end process;

end architecture;
```

Executed using 
`ghdl -c -v --std=08 .\tb_minimal.vhd -r tb_minimal
`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r987.gd45f4a31e) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-w64-mingw32
C:\***\bin\
Command line:
C:\***bin\ghdl.exe -c -v --std=08 .\tb_minimal.vhd -r tb_minimal
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap7.adb:3809
Load address: 0x7ff7d4d10000
Call stack traceback locations:
0x7ff7d50cbfe6 0x7ff7d50cc797 0x7ff7d50ccf96 0x7ff7d50cf24a 0x7ff7d50cf5e6 0x7ff7d50cf66e 0x7ff7d50c957c 0x7ff7d510b9d4 0x7ff7d511b88e 0x7ff7d511b9bb 0x7ff7d511ed20 0x7ff7d5123d34 0x7ff7d50e649e 0x7ff7d50d96ac 0x7ff7d50eabea 0x7ff7d50e5092 0x7ff7d512c866 0x7ff7d5067fc7 0x7ff7d4f7c2d7 0x7ff7d51300db 0x7ff7d4d1363e 0x7ff7d4d112ec 0x7ff7d4d11404 0x7ffa27b426bb 0x7ffa2922dfb6
******************************************************************
```

tgingold, 2022-12-30:
I can reproduce the crash.  For such a concatenation, you can use `&`.

closed 2022-12-31 by tgingold

maleiter, 2023-01-02:
@tgingold Thank you for fixing the bug so quickly!

commit 3285adae (2022-12-31): trans-chap7: handle unbounded expressions in aggregates.

Fix #2288