ghdl/ghdl #2291: Segmentation fault when simulating an file initialized big size array/ram

Pedroo64, 2022-12-29:
**Description**
GHDL causes segmentation fault when trying to simulate a file initialized big size array/ram.

**Expected behaviour**
On others simulators like Modelsim, the design simulates with no problem. On modelsim, it must be the 64 bit version to simulate big size ram.

**How to reproduce?**

memory.vhd
```vhd :file: memory.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

use STD.TEXTIO.all;
use ieee.std_logic_textio.all;

entity memory is
    generic (
        G_DEPTH : integer := 8192;
        G_ADDR_WIDTH : integer := 16;
        G_DATA_WIDTH : integer := 32;
        G_INIT_FILE : string := "none"
    );
    port (
        clk_i : in std_logic;
        addr_i : in std_logic_vector(G_ADDR_WIDTH - 1 downto 0);
        data_i : in std_logic_vector(G_DATA_WIDTH - 1 downto 0);
        write_i : in std_logic;
        valid_i : in std_logic;
        data_o : out std_logic_vector(G_DATA_WIDTH - 1 downto 0)
    );
end entity memory;

architecture rtl of memory is
    type mem_t is array (0 to G_DEPTH - 1) of std_logic_vector(G_DATA_WIDTH - 1 downto 0);
    impure function ocram_readmemfile(file_name : STRING) return mem_t is
        file file_handle : TEXT;
        variable current_line : LINE;
        variable temp_word : STD_LOGIC_VECTOR(G_DATA_WIDTH - 1 downto 0);
        variable result : mem_t := (others => (others => '0'));
    begin
        if file_name /= "none" then
            file_open(file_handle, file_name,  read_mode);
            for i in 0 to Result'length - 1 loop
                exit when endfile(file_handle);
                readline(file_handle, current_line);
                hread(current_line, temp_word);
                result(i) := temp_word;
            end loop;
        end if;
        return result;
    end function;
begin
    
    process (clk_i)
        variable mem : mem_t := ocram_readmemfile(G_INIT_FILE);
    begin
        if rising_edge(clk_i) then
            if valid_i = '1' then
                data_o <= mem(to_integer(unsigned(addr_i)));
                if write_i = '1' then
                    mem(to_integer(unsigned(addr_i))) := data_i;
                end if;
            end if;
        end if;
    end process;
end architecture rtl;
```
tb_mem.vhd
```vhd :file: tb_mem.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity tb_mem is
end entity tb_mem;

architecture rtl of tb_mem is
    constant C_ADDR_WIDTH : integer := 19;
    constant CLK_PER : time := 20 ns;
    signal clk : std_logic := '0';
    signal addr : unsigned(C_ADDR_WIDTH - 1 downto 0) := (others => '0');
begin
    
    clk <= not clk after CLK_PER / 2;
    
    process (clk)
    begin
        if rising_edge(clk) then
            addr <= addr + 1;
        end if;
    end process;

    u_memory : entity work.memory
        generic map (
            G_DEPTH => integer(2**addr'length),
            G_ADDR_WIDTH => addr'length,
            G_DATA_WIDTH => 8,
            G_INIT_FILE => "memory_init.mem"
        )
        port map (
            clk_i => clk,
            addr_i => std_logic_vector(addr),
            data_i => (others => '0'),
            write_i => '0',
            valid_i => '1',
            data_o => open
        );

end architecture rtl;
```

Dummy memory init file (memory_init.mem)
```
00
01
02
03
04
05
06
07
08
09
0A
0B
0C
0D
0E
0F
10
11
12
13
14
15
16
17
18
19
1A
1B
1C
1D
1E
1F
20
21
22
23
24
25
26
27
28
29
2A
2B
2C
2D
2E
2F
30
31
32
33
34
35
36
37
38
39
3A
3B
3C
3D
3E
3F
40
41
42
43
44
45
46
47
48
49
4A
4B
4C
4D
4E
4F
50
51
52
53
54
55
56
57
58
59
5A
5B
5C
5D
5E
5F
60
61
62
63
64
65
66
67
68
69
6A
6B
6C
6D
6E
6F
70
71
72
73
74
75
76
77
78
79
7A
7B
7C
7D
7E
7F
80
81
82
83
84
85
86
87
88
89
8A
8B
8C
8D
8E
8F
90
91
92
93
94
95
96
97
98
99
9A
9B
9C
9D
9E
9F
A0
A1
A2
A3
A4
A5
A6
A7
A8
A9
AA
AB
AC
AD
AE
AF
B0
B1
B2
B3
B4
B5
B6
B7
B8
B9
BA
BB
BC
BD
BE
BF
C0
C1
C2
C3
C4
C5
C6
C7
C8
C9
CA
CB
CC
CD
CE
CF
D0
D1
D2
D3
D4
D5
D6
D7
D8
D9
DA
DB
DC
DD
DE
DF
E0
E1
E2
E3
E4
E5
E6
E7
E8
E9
EA
EB
EC
ED
EE
EF
F0
F1
F2
F3
F4
F5
F6
F7
F8
F9
FA
FB
FC
FD
FE
FF
```
On **tb_mem.vhd** if **C_ADDR_WIDTH** >= 19 it crashes.

This sample project can be downloaded [here](https://github.com/ghdl/ghdl/files/10320538/ghdl_test.tar.gz). Run make under rtl/proj/ghdl.

Compile commands : 
```sh :image: ghdl/ghdl:bullseye-mcode
$ ghdl -i memory.vhd tb_mem.vhd
$ ghdl -m -fsynopsys tb_mem
$ ghdl -r -fsynopsys tb_mem --max-stack-alloc=4096 --stop-time=40us
Segmentation fault
$ ghdl -v
GHDL 3.0.0-dev (2.0.0.r1064.g07b6da32b.dirty) [Dunoon edition]
 Compiled with GNAT Version: 10.4.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

dmesg output : 
```
[ 9807.312339] potentially unexpected fatal signal 11.
[ 9807.313350] CPU: 2 PID: 27534 Comm: ghdl Not tainted 5.15.79.1-microsoft-standard-WSL2 #1
[ 9807.314213] RIP: 0033:0x7f0504075e8d
[ 9807.314582] Code: ff ff 00 00 00 00 8b 85 38 ff ff ff 3b 85 3c ff ff ff 0f 85 05 00 00 00 e9 24 00 00 00 48 8b 85 40 ff ff ff 8b 8d 38 ff ff ff <c6> 04 08 02 8b 85 38 ff ff ff 05 01 00 00 00 89 85 38 ff ff ff eb
[ 9807.316752] RSP: 002b:00007ffe995dc630 EFLAGS: 00010293
[ 9807.317481] RAX: 00007ffe995dc630 RBX: 00007ffe99ddc9e8 RCX: 0000000000000000
[ 9807.318343] RDX: 0000000000000000 RSI: 000055fe6e8dd650 RDI: 0000000000400000
[ 9807.318919] RBP: 00007ffe999dc730 R08: 000055fe6e8dd620 R09: 000055fe6e8dd620
[ 9807.319477] R10: 0000000000000001 R11: 000055fe6dcba038 R12: 000055fe6e0de19d
[ 9807.319872] R13: 000055fe6e0de1a0 R14: 0000000000000001 R15: 0000000000000001
[ 9807.320566] FS:  00007f05042a3740 GS:  0000000000000000
```

**Context**
- OS: 5.15.79.1-microsoft-standard-WSL2
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `07b6da32b4c47519cfe6f02146a3fc099fbcdcbe`

**Additional context**
I've seen similar issues (#1662,  #812,  #1970) and changed the ram type from **signal** to **variable**. When I initialize the ram with **(others => (others => '0'))** and run it doesn't cause segmentation fault. When I initialize with a file content it crashes.

diogratia, 2022-12-30:
With an llvm or gcc version of ghdl:

```none
%: ghdl -r tb_mem --max-stack-alloc=2048
./tb_mem:error: declaration of a too large object (4096 > --max-stack-alloc=2048 KB)
in process .tb_mem(rtl).u_memory@memory(rtl).P0
./tb_mem:error: error during elaboration
%: ghdl -r tb_mem --max-stack-alloc=4096
%:
```

Allow enough stack space and it will simulate. ghdl --run-help on an mcode version shows --max-stack-alloc is supported.

You are only initializing 256 locations out of 8192, it's typical to load a large FPGA memory with a small boot image or small set of commonly used constants. Your statement *"When I initialize the ram with (others => (others => '0')) and run it doesn't cause segmentation fault. When I initialize with a file content it crashes."* provided the idea of using several subtypes of an unconstrained mem_t array type to shorten the expression returned on stack from function ocram_readmemfile.

That worked with an array subtype of 256 words without changing stack allocations:

```none
%: ghdl -a -fsynopsys tb_mem.vhdl
%: ghdl -e -fsynopsys tb_mem
%: ghdl -r tb_mem
%:
```

```vhdl
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

use STD.TEXTIO.all;
use ieee.std_logic_textio.all;

entity memory is
    generic (
        G_DEPTH : integer := 8192;
        G_ADDR_WIDTH : integer := 16;
        G_DATA_WIDTH : integer := 32;
        G_INIT_FILE : string := "none";
        G_INIT_SIZE:  natural := 256  -- CHANGED, ADDED
    );
    port (
        clk_i : in std_logic;
        addr_i : in std_logic_vector(G_ADDR_WIDTH - 1 downto 0);
        data_i : in std_logic_vector(G_DATA_WIDTH - 1 downto 0);
        write_i : in std_logic;
        valid_i : in std_logic;
        data_o : out std_logic_vector(G_DATA_WIDTH - 1 downto 0)
    );
end entity memory;

architecture rtl of memory is
    type mem_t is array (natural range <>) of std_logic_vector(G_DATA_WIDTH - 1 downto 0);
    subtype progd is mem_t (0 to G_INIT_SIZE - 1); 
    subtype initd0 is mem_t (G_INIT_SIZE to G_DEPTH - 1);
    impure function ocram_readmemfile(file_name : STRING) return progd is
        file file_handle : TEXT;
        variable current_line : LINE;
        variable temp_word : STD_LOGIC_VECTOR(G_DATA_WIDTH - 1 downto 0);
        variable result : progd := (others => (others => '0'));
    begin
        if file_name /= "none" then
            file_open(file_handle, file_name,  read_mode);
            for i in result'range loop
                exit when endfile(file_handle);
                readline(file_handle, current_line);
                hread(current_line, temp_word);
                result(i) := temp_word;
            end loop;
        end if;
        return result;
    end function;
begin
    
    process (clk_i)
        variable mem: mem_t(0 to G_DEPTH - 1) := ocram_readmemfile(G_INIT_FILE) & initd0'(others => (others => '0'));
    begin
        if rising_edge(clk_i) then
            if valid_i = '1' then
                data_o <= mem(to_integer(unsigned(addr_i)));
                if write_i = '1' then
                    mem(to_integer(unsigned(addr_i))) := data_i;
                end if;
            end if;
        end if;
    end process;
end architecture rtl;
```

(The testbench was not modified.)

Pedroo64, 2022-12-30:
> Allow enough stack space and it will simulate

On my previous comment, at the compile commands I've allowed 4096 KB of stack.
```
...
$ ghdl -r -fsynopsys tb_mem --max-stack-alloc=4096 --stop-time=40us
Segmentation fault
$ ghdl -v
GHDL 3.0.0-dev (2.0.0.r1064.g07b6da32b.dirty) [Dunoon edition]
 Compiled with GNAT Version: 10.4.0
 mcode code generator
Written by Tristan Gingold.
...
```
I've tried the changes you have made on **memory.vhd** and still causes a segfault with mcode backend and 255 (return code) with llvm backend with the same messages on dmesg. I'm wondering if the problem is not WSL... Should try to install the Windows version but there is no instructions how to build it... 🙁

> You are only initializing 256 locations out of 8192, it's typical to load a large FPGA memory with a small boot image or small set of commonly used constants. Your statement "When I initialize the [...]

I understand your point but this was only an example to show that if the memory is > 2^18 or > 256 KB it causes a seg fault and I have some tests that the image is bigger than 256 KB.

Pedroo64, 2022-12-30:
I've tried to build GHDL on Windows which is my host system using MSYS2. I could simulate until C_ADDR_WIDTH <= 20 (1 MB). If C_ADDR_WIDTH  >= 21 it causes a segmentation fault with mcode back end and 127 (return code) with llvm back end. I'm I missing free RAM memory? I'm saying this because my host system has 16 GB and WSL2 has 8 GB. But still I've rebooted my system with an bunch of free RAM but still it crashes so I don't know if this is the issue...

diogratia, 2022-12-31:
You missed some additional context. See  issue #1592 as well.

My immediate interest is in finding a work around. Without knowing your application (device or system) we can look at variations of your impure function to provide initial values.

So I tried returning pages from ocram_readmemfile:

```vhdl
architecture pages of memory is
    constant PAGESIZE: natural := 16384;
    use ieee.math_real.all;  -- ceil
    constant NUM_PAGES: natural := natural(ceil(real(G_DEPTH)/real(PAGESIZE)));
    type mem_t is array (natural range <>) of 
                        std_logic_vector(G_DATA_WIDTH - 1 downto 0);

    impure function ocram_readmemfile(file_name : STRING; page: natural) 
            return mem_t is
        file file_handle : TEXT;
        variable current_line : LINE;
        variable temp_word : STD_LOGIC_VECTOR(G_DATA_WIDTH - 1 downto 0);
        variable result: 
            mem_t (page * PAGESIZE to (page * PAGESIZE + PAGESIZE - 1))
                 := (others => (others => '0'));
    begin
        report "page " & integer'image(page) & " of " 
                       & integer'image(NUM_PAGES) & " pages";
        if file_name /= "none" then
            file_open(file_handle, file_name,  read_mode);
            for i in 0 to G_DEPTH - 1 loop  
                exit when endfile(file_handle);
                if i >= result'left and i <= result'right then
                    readline(file_handle, current_line);
                    hread(current_line, temp_word);
                    result(i) := temp_word;
                end if;
                if i = result'right then
                    exit;
                end if;
            end loop;
            file_close (file_handle);
        end if;
        return result;
    end function;
begin

    process (clk_i)
        variable mem: mem_t(0 to G_DEPTH - 1) := 
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 0) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 1) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 2) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 3) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 4) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 5) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 6) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 7) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 8) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 9) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 10) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 11) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 12) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 13) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 14) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 15) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 16) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 17) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 18) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 19) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 20) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 21) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 22) &
                      ocram_readmemfile(file_name => G_INIT_FILE, page => 23) &
[... 117 lines cut ...]
                    exit when endfile (f);
                    readline (f, l);
                    hread (l, temp_word);  -- from std_logic_textio
                    mem(i) := temp_word;
                end loop;
                file_close (f);
            end if;
            inited := TRUE;
        end if;
        if rising_edge(clk_i) then
            if valid_i = '1' then
                data_o <= mem(to_integer(unsigned(addr_i)));
                if write_i = '1' then
                    mem(to_integer(unsigned(addr_i))) := data_i;
                end if;
            end if;
        end if;
    end process;
end architecture proc;
```
This doesn't rely on the return value stack allocation size.

This would be useful where the initial value of mem isn't the only method of conveying any bit image data to an FPGA synthesis tool.

I'll leave it to Tristan to address ghdl changes.

Pedroo64, 2023-01-02:
I've tried your second approach by adding a reset signal that initializes the memory with the file content when it's asserted.
```vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

use STD.TEXTIO.all;
use ieee.std_logic_textio.all;

entity memory is
    generic (
        G_DEPTH : integer := 8192;
        G_ADDR_WIDTH : integer := 16;
        G_DATA_WIDTH : integer := 32;
        G_INIT_FILE : string := "none"
    );
    port (
        arst_i : in std_logic;
        clk_i : in std_logic;
        addr_i : in std_logic_vector(G_ADDR_WIDTH - 1 downto 0);
        data_i : in std_logic_vector(G_DATA_WIDTH - 1 downto 0);
        write_i : in std_logic;
        valid_i : in std_logic;
        data_o : out std_logic_vector(G_DATA_WIDTH - 1 downto 0)
    );
end entity memory;

architecture rtl of memory is
    type mem_t is array (0 to G_DEPTH - 1) of std_logic_vector(G_DATA_WIDTH - 1 downto 0);
begin
    process (clk_i, arst_i)
        variable mem : mem_t;
        procedure ocram_readmemfile(file_name : string) is
            file file_handle : text;
            variable current_line : line;
            variable temp_word : std_logic_vector(G_DATA_WIDTH - 1 downto 0);
        begin
            if file_name /= "none" then
                file_open(file_handle, file_name,  read_mode);
            end if;
            for i in 0 to G_DEPTH - 1 loop
                if file_name /= "none" and not endfile(file_handle) then
                    readline(file_handle, current_line);
                    hread(current_line, temp_word);
                else
                    temp_word := (others => '0');
                end if;
                mem(i) := temp_word;
            end loop;
            if file_name /= "none" then
                file_close(file_handle);
            end if;
        end procedure;
    begin
        if arst_i = '1' then
            ocram_readmemfile(G_INIT_FILE);
        elsif rising_edge(clk_i) then
            if valid_i = '1' then
                data_o <= mem(to_integer(unsigned(addr_i)));
                if write_i = '1' then
                    mem(to_integer(unsigned(addr_i))) := data_i;
                end if;
            end if;
        end if;
    end process;
end architecture rtl;
```
On the testbench we need to connect and assert and deassert the reset.
```vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity tb_mem is
end entity tb_mem;

architecture rtl of tb_mem is
    constant C_ADDR_WIDTH : integer := 24;
    constant CLK_PER : time := 20 ns;
    signal arst, clk : std_logic := '0';
    signal addr : unsigned(C_ADDR_WIDTH - 1 downto 0) := (others => '0');
begin
    
    arst <= '1', '0' after 63 ns;
    clk <= not clk after CLK_PER / 2;
    
    process (clk, arst)
    begin
        if arst = '1' then
            addr <= (others => '0');
        elsif rising_edge(clk) then
            addr <= addr + 1;
        end if;
    end process;

    u_memory : entity work.memory
        generic map (
            G_DEPTH => integer(2**addr'length),
            G_ADDR_WIDTH => addr'length,
            G_DATA_WIDTH => 8,
            G_INIT_FILE => "memory_init.mem"
        )
        port map (
            arst_i => arst,
            clk_i => clk,
            addr_i => std_logic_vector(addr),
            data_i => (others => '0'),
            write_i => '0',
            valid_i => '1',
            data_o => open
        );
end architecture rtl;
```
With your approach, no need to increase the stack. Also we can change **mem** from **variable** to **signal** which that also doesn't cause a segmentation fault. Still with **mem** as a **variable** performance is better at initialization.