ghdl/ghdl #2296: Exception STORAGE_ERROR with ghdl-mcode

marzoul, 2023-01-04:
**Description**
ghdl-mcode crashes with STORAGE_ERROR exception when running a very tiny simulation.
The issue happens with `use ieee.numeric_std.all` and disappears without.
The issue does not happen with ghdl-llvm.

**Expected behaviour**
The simulation runs with no error.

**How to reproduce?**
Use ghdl-mcode to reproduce the issue. Tested with ghdl commit `45820b4ee0b61090a5f950da1485b2c635eb6396` from 2023-01-03.
Tested on an up-to-date archlinux distribution.

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ghdl_crash is
end ghdl_crash;

architecture simu of ghdl_crash is
	signal s : std_logic;
begin
	s <= '0';
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

**Context**
Please, provide the following information:

- OS: Archlinux, AUR package
- Origin: Built from sources: `commit 45820b4ee0b61090a5f950da1485b2c635eb6396`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r1105.g45820b4ee) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-pc-linux-gnu
/home/prostboa/Projets/projet-esprit/vc709/simu/simu-nnlayer-recode-to-ter/ghdl-crash/
Command line:
ghdl --elab-run ghdl_crash
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x55a98304ce0f 0x7fd440e979fe 0x55a982d9f27f 0x55a982dc1a32 0x55a982dc6eec 0x55a982dc750b 0x55a982e50327 0x55a982fb3fd1 0x55a982fb70d9 0x55a982fe61cc 0x55a982ff2235 0x55a982ff2f35 0x55a982ff22f3 0x55a982ff3765 0x55a982fd4d14 0x55a982fe04ca 0x55a982fd6073 0x55a982fcb3d7 0x55a982fd0ea8 0x55a9830001b4 0x55a982f78740 0x55a982ed38f8 0x55a9830022c9 0x55a982d310fd 0x7fd440e8228e 0x7fd440e82348 0x55a982d31173 0xfffffffffffffffe
******************************************************************
```

**Additional context**
Just commenting the VHDL line `use ieee.numeric_std.all;` makes the simulation run with no error.
It may be possible that the issue was made visible by the update of the Archlinux system (11 days ago). Because previously-built ghdl-mcode packages on commits from 2022-12-03 and 2022-09-07 exhibit the same erroneous behaviour now, but used to work with no error in 2022.

tgingold, 2023-01-04:
It's weird as this is a very simple design.

Can you try to run under gdb and display the backtrace ?

$ gdb --args ghdl --elab-run ghdl_crash
...
(gdb) run
...
(gdb) bt

marzoul, 2023-01-04:
I rebuilt the ghdl-mcode package with debug stuff. Here is the backtrace :

```
Program received signal SIGSEGV, Segmentation fault.
0x00005555555bb621 in binary_file__gen_8 (b=83) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/ortho/mcode/binary_file.adb:365
warning: Source file is more recent than executable.
365	      Cur_Sect.Data (Cur_Sect.Pc) := B;
(gdb) bt
#0  0x00005555555bb621 in binary_file__gen_8 (b=83) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/ortho/mcode/binary_file.adb:365
#1  0x00005555555dfa80 in ortho_code__x86__emits__emit_const (val=7793) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/ortho/mcode/ortho_code-x86-emits.adb:3300
#2  0x00005555555c5705 in ortho_code__x86__emits__emit_init_value (val=7793, decl=1470) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/ortho/mcode/ortho_code-x86-emits.adb:3343
#3  ortho_code__x86__abi__expand_init_value (val=7793, decl=1470) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:233
#4  ortho_code__decls__new_init_value (decl=1470, val=7793) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/ortho/mcode/ortho_code-decls.adb:362
#5  0x000055555565284d in ortho_mcode__finish_init_value (val=7793, decl=1470) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/ortho/mcode/ortho_mcode.adb:142
#6  trans__chap10__create_global_const (name=1910, vtype=1486, storage=<optimized out>, initial_value=7793) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans.adb:502
#7  0x00005555557de0c5 in trans__chap7__translate_static_implicit_conv (expr=expr@entry=7793, expr_type=expr_type@entry=5596, res_type=res_type@entry=672) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans-chap7.adb:93
#8  0x00005555557e047b in trans__chap7__translate_composite_literal (str=5595, res_type=672) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans-chap7.adb:517
#9  0x000055555582fdbd in trans__chap8__translate_report (stmt=3810, subprg=627, level=364) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans-chap8.adb:1267
#10 0x000055555583fa06 in trans__chap8__translate_assertion_statement (stmt=3810) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans-chap8.adb:1323
#11 trans__chap8__translate_statement (stmt=3810) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans-chap8.adb:4997
#12 0x0000555555840737 in trans__chap8__translate_statements_chain (first=<optimized out>) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans-chap8.adb:5082
#13 0x000055555582c866 in trans__chap8__translate_if_statement_direct (stmt=3804) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans-chap8.adb:376
#14 0x000055555583e395 in trans__chap8__translate_statement (stmt=3804) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans-chap8.adb:5025
#15 0x000055555580ad58 in trans__chap8__translate_statements_chain_has_return (first=<optimized out>) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans-chap8.adb:5095
#16 trans__chap2__translate_subprogram_body (subprg=3776) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans-chap2.adb:681
#17 0x000055555581eb3f in trans__chap4__translate_declaration_chain_subprograms (parent=<optimized out>, what=trans__subprg_translate_spec_and_body) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans-chap4.adb:2611
#18 0x000055555580cb5e in trans__chap2__translate_package_body (bod=832) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans-chap2.adb:979
#19 0x00005555557fbdb1 in translation__translate (unit=<optimized out>, main=<optimized out>) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/translation.adb:304
#20 0x000055555584f0c8 in trans__chap12__elaborate (whole=true, config=<optimized out>) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/trans-chap12.adb:625
#21 translation__elaborate (whole=true, config=<optimized out>) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/translate/translation.adb:2231
#22 ghdlrun__compile_elab (cmd_name=..., args=...) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/ghdldrv/ghdlrun.adb:206
#23 0x000055555578c991 in ghdlcomp__perform_action__3 (cmd=..., args=...) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/ghdldrv/ghdlcomp.adb:144
#24 0x00005555556c9dd6 in ghdlmain__main () at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/ghdldrv/ghdlmain.adb:518
#25 0x0000555555572e37 in _ada_ghdl_jit () at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/ghdldrv/ghdl_jit.adb:40
#26 main (argc=argc@entry=3, argv=argv@entry=0x7fffffffe648, envp=<optimized out>) at /home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/b~ghdl_jit.adb:1044
#27 0x00007ffff7cbb290 in __libc_start_call_main (main=main@entry=0x555555572090 <main>, argc=argc@entry=3, argv=argv@entry=0x7fffffffe648) at ../sysdeps/nptl/libc_start_call_main.h:58
#28 0x00007ffff7cbb34a in __libc_start_main_impl (main=0x555555572090 <main>, argc=3, argv=0x7fffffffe648, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffe638) at ../csu/libc-start.c:381
#29 0x0000555555572ec5 in _start () at ../sysdeps/x86_64/start.S:115
(gdb)
```

Note that the compilation of ghdl-mcode emits a lot of warnings, some of them are quite... dubious.
[... cut ...]
/home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/bug.adb:29:4: warning: type of ‘bug__gnat_version’ does not match original declaration [-Wlto-type-mismatch]
   29 |    GNAT_Version : constant String (1 .. 31 + 15);
      |    ^
/home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/libghdl/libghdl.adb:133:4: note: type ‘character[1:25]’ should match type ‘character[1:46]’
  133 |    Gnat_Version : constant String := "unknown compiler version" & ASCII.NUL;
      |    ^
/home/prostboa/Dev/AUR/ghdl-mcode-git/src/ghdl/src/vhdl/libghdl/libghdl.adb:133:4: note: ‘libghdl__gnat_version’ was previously declared here

[...]
```

marzoul, 2023-01-04:
Additional note : Under valgrind, there is no crash and the simulation runs with no error, not even a warning (nice!).
Given that valgring impacts memory allocation and in particular memory reuse, the root cause of the issue is probably linked to manipulation of memory, or to code behavior that depends on comparison of pointers to memory locations, that sort of things.

tgingold, 2023-01-05:
Interesting.  This might be an issue due to gcc 12.  I will try to build and use it.

marzoul, 2023-01-06:
Update : after a system update this morning, the issue is no longer happening.

There was no update related to C compiler, C libraries or to ada.
Only meaningful candidate is the update of linux kernel and libsigsegv. Still a bit weird, but why not.

I also recompiled ghdl-mcode after the update. With today's code and the one from 2 days ago.
The compilation warnings are no longer present (those related to compilation of ghdl itself only).

The issue just vanished. Probably the laptop has been hit by a solar proton somewhere... a very selective one...

closed 2023-01-06 by marzoul

tgingold, 2023-01-06:
Weird indeed.  I was able to build ghdl with gcc 12.2, and it passes the tests.