ghdl/ghdl #2297: build error on CentOs 7.9.2009 with gcc backend

stdefeber, 2023-01-05:
**Description**

I am trying to build ghdl on CentOs 7 with gcc back-end. In the past I have successfully build ghdl on other systems.
Initially I tried with gcc source 11.3.0. Which failed. Next, I tried with gcc version 4.9.4 to see whether that would fit this "outdated" system better.

The build exits with the following error:

`gcc -c -I./ -I/data/home/simonthijs.defeber@axign.local/temp/gcc-4.9.4/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /data/home/simonthijs.defeber@axign.local/temp/ghdl_build/ghdl/build/gcc-objs/gcc/vhdl/trans-chap3.o /data/home/simont/temp/gcc-4.9.4/gcc/vhdl/trans-chap3.adb
trans-chap3.adb:1221:12: statement expected
gnatmake: "/data/home/simonthijs.defeber@axign.local/temp/gcc-4.9.4/gcc/vhdl/trans-chap3.adb" compilation error
make[2]: *** [ghdl1] Error 4
make[2]: *** Waiting for unfinished jobs....`

**Context**
Please, provide the following information:

- OS: CentOs 7.9.2009 (Core)
- Origin:
  - [ ] Built from sources: 8185ed878
  - [ ] Built from gcc source: 11.3.0
  - [ ] Built from gcc source: 4.9.4

tgingold, 2023-01-05:
I will fix it, but you just need to add 'null;' before line 1223.

What is the issue with gcc 11.3 ?

stdefeber, 2023-01-06:
I guess it is the same error.

' Searching /usr/include/./phonon/Phonon
gcc -c -I./ -I/data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /data/home/simonthijs.defeber@axign.local/temp/ghdl_build/ghdl/build/gcc-objs/gcc/vhdl/vhdl-lists.o /data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl/vhdl-lists.ads
gcc -c -I./ -I/data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /data/home/simonthijs.defeber@axign.local/temp/ghdl_build/ghdl/build/gcc-objs/gcc/vhdl/vhdl-parse.o /data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl/vhdl-parse.adb
gcc -c -I./ -I/data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /data/home/simonthijs.defeber@axign.local/temp/ghdl_build/ghdl/build/gcc-objs/gcc/vhdl/ortho_nodes.o /data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl/ortho_nodes.ads
gcc -c -I./ -I/data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /data/home/simonthijs.defeber@axign.local/temp/ghdl_build/ghdl/build/gcc-objs/gcc/vhdl/trans.o /data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl/trans.adb
gcc -c -I./ -I/data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /data/home/simonthijs.defeber@axign.local/temp/ghdl_build/ghdl/build/gcc-objs/gcc/vhdl/trans-chap1.o /data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl/trans-chap1.adb
gcc -c -I./ -I/data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /data/home/simonthijs.defeber@axign.local/temp/ghdl_build/ghdl/build/gcc-objs/gcc/vhdl/trans-chap12.o /data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl/trans-chap12.adb
gcc -c -I./ -I/data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /data/home/simonthijs.defeber@axign.local/temp/ghdl_build/ghdl/build/gcc-objs/gcc/vhdl/trans-chap2.o /data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl/trans-chap2.adb
gcc -c -I./ -I/data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /data/home/simonthijs.defeber@axign.local/temp/ghdl_build/ghdl/build/gcc-objs/gcc/vhdl/trans-chap3.o /data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl/trans-chap3.adb
trans-chap3.adb:1221:12: statement expected
gnatmake: "/data/home/simonthijs.defeber@axign.local/temp/gcc-11.3.0/gcc/vhdl/trans-chap3.adb" compilation error
make[2]: *** [ghdl1] Error 4
make[2]: *** Waiting for unfinished jobs....
Making symbolic directory links
sed: -e expression #1, char 100: unknown option to `s'
Fixing directory /usr/include into /data/home/simonthijs.defeber@axign.local/temp/ghdl_build/ghdl/build/gcc-objs/gcc/include-fixed'

stdefeber, 2023-01-06:
>> I will fix it, but you just need to add 'null;' before line 1223.

@tgingold, can you please post the (full) code change ? I am not proficient in ADA :(
I am willing to rebuild it on both gcc versions.

tgingold, 2023-01-06:
I have pushed the fix.  Can you pull again and rebuild ?

stdefeber, 2023-01-07:
Rebuild with gcc-11.3.0.
The earlier error is gone. However another one showed up.

Also tried with --disbale-synth option. Did not work as well.

`gcc -c -I./ -I/data/home/<user>/temp/gcc-11.3.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /data/home/<user>/temp/ghdl_build/ghdl/build/gcc-objs/gcc/vhdl/synth-vhdl_expr.o /data/home/<user>/temp/gcc-11.3.0/gcc/vhdl/synth-vhdl_expr.adb
gcc -c -I./ -I/data/home/<user>/temp/gcc-11.3.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /data/home/<user>/temp/ghdl_build/ghdl/build/gcc-objs/gcc/vhdl/elab-vhdl_context.o /data/home/<user>/temp/gcc-11.3.0/gcc/vhdl/elab-vhdl_context.adb
gcc -c -I./ -I/data/home/<user>/temp/gcc-11.3.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /data/home/s<user>/temp/ghdl_build/ghdl/build/gcc-objs/gcc/vhdl/grt-arith.o /data/home/<user>/temp/gcc-11.3.0/gcc/vhdl/grt-arith.adb
gcc -c -I./ -I/data/home/<user>/temp/gcc-11.3.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC -I- -o /data/home/<user>/temp/ghdl_build/ghdl/build/gcc-objs/gcc/vhdl/netlists.o /data/home/<user>/temp/gcc-11.3.0/gcc/vhdl/netlists.adb
netlists.adb:935:13: iterator is an Ada 2012 feature
gnatmake: "/data/home/<user>/temp/gcc-11.3.0/gcc/vhdl/netlists.adb" compilation error
make[2]: *** [ghdl1] Error 4
rm gcc.pod
make[2]: Leaving directory `/data/home/<user>/temp/ghdl_build/ghdl/build/gcc-objs/gcc'
make[1]: *** [all-gcc] Error 2
make[1]: Leaving directory `/data/home/<user>/temp/ghdl_build/ghdl/build/gcc-objs'
make: *** [all] Error 2`

tgingold, 2023-01-08:
Probably your gcc compiler is older than expected.

Can you append '-gnat12' to 'GHDL_ADAFLAGS' in `scripts/gcc/Make-lang.in` ?

stdefeber, 2023-01-09:
I found that a newer version of gcc is avilable on centos7.9.
Being gcc 11.2.1-9.

So I installed via:

>sudo yum install devtoolset-11-gcc.x86_64

And by the following command I made it available in a new bash environment

> sudo scl enable devtoolset-11 bash

I still needed to add  '-gnat12'.
And it agian failed with the same error as in the previous comment.

tgingold, 2023-01-09:
I am pretty sure you are still using an old ada compiler.  What are the outputs of `gnatls -v` ?

stdefeber, 2023-01-09:
Thanks. You are right.
I need to find out whether a newer ada compiler is available.

stdefeber, 2023-01-12:
Almost.......

I uninstalled gnat 4.8.5 and installed the ADA compiler from https://www.adacore.com.
Compilation now fails in the final step.

`/bin/sh /data/home/<user>/temp/gcc-11.3.0/gcc/../move-if-change tmp-fixinc_list fixinc_list
echo timestamp > s-fixinc_list
if [ -f ../stage_final ] \
   && cmp -s ../stage_current ../stage_final; then \
  cp ../prev-gcc/cc1-checksum.c cc1-checksum.c; \
else \
  build/genchecksum c/c-lang.o c-family/stub-objc.o attribs.o c/c-errors.o c/c-decl.o c/c-typeck.o c/c-convert.o c/c-aux-info.o c/c-objc-common.o c/c-parser.o c/c-fold.o c/gimple-parser.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-dump.o .o c-family/c-lex.o c-family/c-omp.o c-family/c-opts.o c-family/c-pch.o c-family/c-ppoutput.o c-family/c-pragma.o c-family/c-pretty-print.o c-family/c-semantics.o c-family/c-ada-spec.o c-family/c-ubsan.o c-family/known-headers.o c-family/c-attribs.o backend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a libcommon.a ../libcpp/libcpp.a ../libiberty/libiberty.a   ../libdecnumber/libdecnumber.a ../libbacktrace/.libs/libbacktrace.a \
                     checksum-options > cc1-checksum.c.tmp &&            \
  /data/home/<user>/temp/gcc-11.3.0/gcc/../move-if-change cc1-checksum.c.tmp cc1-checksum.c; \
fi
no -c -aI/data/home/<user>/temp/gcc-11.3.0/gcc/vhdl --subdirs=vhdl ortho_gcc-main \
 -cargs -g -O2 -gnata -gnaty3befhkmr -gnatwa -gnatwC
make[2]: no: Command not found
make[2]: *** [ghdl1] Error 127
make[2]: *** Waiting for unfinished jobs....
rm gcc.pod`

stdefeber, 2023-01-16:
The configure log file gave away the details.
Although the correct version of gcc was used, it did not so for c++ or g++.
After resolving this issue the build passed.

closed 2023-01-16 by stdefeber