ghdl/ghdl #2299: Crash when printing undefined real signals

bjourne, 2023-01-06:
See this MWE:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.textio.all;

entity tb_dct8 is
end tb_dct8;

architecture beh of tb_dct8 is
    procedure write_arr(x : real_vector) is
        variable line0 : line;
    begin
        for n in x'range loop
            write(line0, to_string(x(n), 4), right, 5);
            write(line0, string'(" "));
        end loop;
        writeline(output, line0);
    end procedure;
    signal y : real_vector(0 to 7);
begin
    process
    begin
        write_arr(y);
        assert false report "all tests passed" severity note;
        wait;
    end process;
end architecture;
```
It produces the following segfault:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r1120.g456c28b50) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-pc-linux-gnu
/home/bjorn/pubwork/hw-design-libs/
Command line:
/home/bjorn/pubwork/ghdl/ghdl_mcode --elab-run -P./build --workdir=build --std=08 tb_dct8
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-images.adb:180 range check failed
Call stack traceback locations:
0x55df3c4dd25e 0x7f861bf047dc 0x7f861bf04aa4 0x55df3c4bcbe5 0x55df3c4bce9e 0x55df3c4bd5af 0x55df3c4bd8dc 0x55df3c45366f 0x55df3c5ce2a7 0x55df3c5ce37e 0x55df3c8b406b 0x55df3c7db4e5 0x55df3c6e12da 0x55df3c8b4b60 0x55df3c453216 0x7f861c11628e 0x7f861c116348 0x55df3c451713 0xfffffffffffffffe
******************************************************************
```

bjourne, 2023-01-06:
Actually, might be something with to_string(). This crashes in the same way:
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.textio.all;

entity tb_dct8 is
end tb_dct8;

architecture beh of tb_dct8 is
    signal r : real;
begin
    process
        variable line0 : line;
    begin
        r <= 123.0;
        write(line0, to_string(r, 8));
        writeline(output, line0);
        wait;
    end process;
end architecture;
```

tgingold, 2023-01-06:
Yes, I can reproduce the crash.

tgingold, 2023-01-06:
You are using to_string() with the digits arguments.  This creates very long strings.  I am testing a fix for that.

commit 0c134f79 (2023-01-06): grt-images.adb: handle very long strings.  Fix #2299
commit cb7a8ebe (2023-01-10): synth: also fix #2299
closed 2023-01-06 by tgingold