ghdl/ghdl #2300: VPI: cannot read constant arrays

Scafir, 2023-01-06:
**Description**
Using VPI (cocotb 0.2.3) with latest ghdl, trying to access a constant array results in `vpi_get: unknown property`.

**Expected behaviour**
The value of the constant array should be accessible and not error should happen.

**How to reproduce?**
```vhd :file: main.vhd
LIBRARY IEEE;
USE IEEE.STD_LOGIC_1164.ALL;
USE ieee.numeric_std.ALL;

ENTITY main IS
    PORT (
        data_out : out  std_logic_vector(15 DOWNTO 0));
END main;

ARCHITECTURE structure OF main IS
    TYPE memT IS ARRAY(0 to 2) OF unsigned(15 DOWNTO 0);
    SIGNAL mem : memT;
    CONSTANT memC : memT := (others => (others => '0'));

BEGIN
    data_out <= std_logic_vector(mem(0));
END structure;
```
```py :file: main_test.py
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
async def init_test(dut):
await Timer(1)

print("Elements in dut:")
print(dir(dut))
print ("")
print("Memory: ")
print(dut.mem[0].value)
# Following line makes GHDL crash
print(dut.memC[0].value)
```

```makefile :file: Makefile
VHDL_SOURCES = $(PWD)/main.vhd
TOPLEVEL = main
MODULE = test_main
SIM = ghdl

include $(shell cocotb-config --makefiles)/Makefile.sim
```

**Context**
Please, provide the following information:

- OS: Fedora 37
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `65a40a067d1b153265eaf8bcadbc225c5eeac041`

**Maybe helpful material**
Issue #237, closed with https://github.com/ghdl/ghdl/commit/c64e7ba56376b933cb0ddb2b4949a83c926a3c67

Scafir, 2023-03-24:
Interestingly, it seems that trying to access a constant inside a nested entity results in the same behaviour:
```vhdl
-- main.vhd
LIBRARY IEEE;
USE IEEE.STD_LOGIC_1164.ALL;

ENTITY sub IS
    PORT (
        data_out : out  std_logic_vector(15 DOWNTO 0)
    );
END sub;

ARCHITECTURE structure OF sub IS
    CONSTANT myConstant : std_logic_vector(15 DOWNTO 0) := X"DEAD";
    SIGNAL mySignal : std_logic_vector(15 downto 0) := X"BEEF";
BEGIN
    data_out <= myConstant;
END structure;

LIBRARY IEEE;
USE IEEE.STD_LOGIC_1164.ALL;

ENTITY main IS
    PORT (
        data_out : out  std_logic_vector(15 DOWNTO 0));
END main;

ARCHITECTURE structure OF main IS
    COMPONENT sub
        port (
            data_out : out  std_logic_vector(15 DOWNTO 0)
        );
    END COMPONENT;
BEGIN
    SUB_E : sub
    port map (
        data_out => data_out
    );
END structure;
```
```Makefile
# Makefile
VHDL_SOURCES = $(PWD)/main.vhd
TOPLEVEL = main
MODULE = test_main
SIM = ghdl

include $(shell cocotb-config --makefiles)/Makefile.sim
```
```py
# test_main.py
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
async def init_test(dut):
    await Timer(1)

    print("Elements in sub:")
    print(dir(dut.sub_e))
    print ("")
    print("Access nested signal")
    print(dut.sub_e.mySignal)

    print("Access nested constant")
    print(dut.sub_e.myConstant)
```

Scafir, 2023-04-06:
With further inspection, it seems any array-based variable is impacted by this. On top of my head:
* std_logic_vector
* signed/unsigned
* arrays
* ufixed/sfixed
* ...

I will delete my previous comment, which is misleading.

umarcor, 2023-04-19:
@Scafir you might find the following references interesting:

- #237
- #1249
- #1297
- #1299
- #1306