ghdl/ghdl #2302: Simple VUnit test fail with latest GHDL build

LarsAsplund, 2023-01-06:
**Description**
Simple VUnit test fail with latest GHDL build

**Expected behaviour**
When running a simple Hello World VUnit example test like `<VUnit install dir>\examples\vhdl\user_guide\tb_example.vhd` the expected behavior is a passing test with the following output

> C:\github\vunit\examples\vhdl\user_guide\tb_example.vhd:19:5:@0ms:(report note): > Hello world!
> simulation stopped @0ms with status 0

This works with GHDL mcode up to at least version:

> GHDL 3.0.0-dev (2.0.0.r600.gcdcbb258b) [Dunoon edition]
> Compiled with GNAT Version: 12.1.0
> mcode code generator

However, it no longer works with nightly. Instead the test fails. The output from the
simulation is the same. The only visible difference is the return code 0xc0000005. The nightly version was

> GHDL 3.0.0-dev (2.0.0.r1130.g7de3e6802) [Dunoon edition]
> Compiled with GNAT Version: 12.2.0
> mcode code generator

tgingold, 2023-01-07:
Is it a windows only issue or does it also happens on Linux ?

LarsAsplund, 2023-01-07:
Yes, only on Windows. Same version with "ubuntu-latest" works fine.

umarcor, 2023-03-07:
I'm closing as invalid, since it seems that "mcode 64 bits" was used on Windows, which is known not to work for simulation but provided for pyGHDL only. See https://github.com/VUnit/vunit/issues/882.

closed 2023-03-07 by umarcor