ghdl/ghdl #2303: Segmentation fault when custom integer type is out of range

kraigher, 2023-01-08:
**Description**
Segmentation fault

**Expected behaviour**
No segmentation fault

**How to reproduce?**
```vhd :file: ent.vhd
entity ent is
end entity;

architecture  a of ent is
  type type_t is range 0 to 8;
  constant foo : type_t := 3 ** 2;
begin
end architecture;
```

```
$ ../ghdl/bin/ghdl.exe --version
GHDL 3.0.0-dev (2.0.0.r950.g674246f83) [Dunoon edition]
 Compiled with GNAT Version: 12.2.0
 mcode code generator
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 ent.vhd
ghdl --elab-run --std=08  ent
```

```
$ ../ghdl/bin/ghdl --elab-run --std=08  ent
D:\repo\ghdl\bin\ghdl.exe:error: bound check failure at test.vhd:6
Segmentation fault
```

tgingold, 2023-01-08:
Probably an issue with the unwinder when computing the backtrace.

tgingold, 2023-01-11:
I was not able to reproduce the segfault.  I will add a test and see how CI deals with it.

Paebbels, 2024-12-19:
@tgingold any actions in this one?