ghdl/ghdl #2304: Building GHDL-GCC on Debian Bullseye (11)

jrtchapple, 2023-01-09:
Below is a script of console interactions attempting to build GHDL with a GCC back end.  I am running Debian Bullseye (11).  As GCC is not included in the base installation, I had to install it as well as gnat from the Debian repository.  Same with libisl-dev, libgmp-dev, libmpfr-dev, and libmpc-dev.

john@debian-desktop:~/ghdl$ mkdir build
john@debian-desktop:~/ghdl$ cd build
john@debian-desktop:~/ghdl/build$ apt list | grep -i installed >t

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

john@debian-desktop:~/ghdl/build$ grep -i libgmp t

libgmp-dev/stable,now 2:6.2.1+dfsg-1+deb11u1 amd64 [installed]
libgmp10/stable,now 2:6.2.1+dfsg-1+deb11u1 amd64 [installed]
libgmpxx4ldbl/stable,now 2:6.2.1+dfsg-1+deb11u1 amd64 [installed,automatic]

john@debian-desktop:~/ghdl/build$ grep -i libisl t

libisl-dev/stable,now 0.23-1 amd64 [installed]
libisl23/stable,now 0.23-1 amd64 [installed,automatic]

john@debian-desktop:~/ghdl/build$ grep -i libmpfr t

libmpfr-dev/stable,now 4.1.0-3 amd64 [installed]
libmpfr6/stable,now 4.1.0-3 amd64 [installed]

john@debian-desktop:~/ghdl/build$ grep -i libmpc t

libmpc-dev/stable,now 1.2.0-1 amd64 [installed]
libmpc3/stable,now 1.2.0-1 amd64 [installed]
libmpcdec6/stable,now 2:0.1~r495-2 amd64 [installed,automatic]

john@debian-desktop:~/ghdl/build$ ../configure --with-gcc=/usr/src/gcc-10/gcc-10.2.0 --prefix=/usr/local >ghdl-config
john@debian-desktop:~/ghdl/build$ make copy-sources >copy-sources-stdout
john@debian-desktop:~/ghdl/build$ mkdir gcc-objs
john@debian-desktop:~/ghdl/build$ cd gcc-objs
[files.tar.gz](https://github.com/ghdl/ghdl/files/10375965/files.tar.gz)

john@debian-desktop:~/ghdl/build/gcc-objs$ /usr/src/gcc-10/gcc-10.2.0/configure --prefix=/usr/local --with-isl-include=/usr/local/include --enable-languages=c,vhdl \
--disable-bootstrap --disable-lto --disable-multilib --disable-libssp \
--disable-libgomp --disable-libquadmath --enable-default-pie >gcc-config-stdout
*** This configuration is not supported in the following subdirectories:
     target-libquadmath gnattools gotools target-libada target-libhsail-rt target-libstdc++-v3 target-libphobos target-zlib target-libbacktrace target-libgfortran target-libgo target-libffi target-libobjc target-libgomp target-liboffloadmic target-libssp target-libitm target-libsanitizer target-libvtv
    (Any other directories should still work fine.)
john@debian-desktop:~/ghdl/build/gcc-objs$

--with-isl-include=/usr/local/include resolves a warning about using in-tree isl.  It does not resolve the lack of support in the above listed subdirectories.

A make command to build GHDL fails.  Are there any more packages that need to be installed?

The attached zipped tarball contains console messages (mostly stderr output) and stdout messages redirected to a file for ghdl-config, make copy-sources, gcc-config, and building ghdl with make.  I haven't gone any further at this time.

Please ignore the strikeout.  Some gremlins added it.

tgingold, 2023-01-09:
Did you try to also enable c++ ?

jrtchapple, 2023-01-09:
Changing

 --enable-languages=c,vhdl 

to

--enable-languages=c,c++,vhdl

had no obvious effect on the build results.  I get the same message from genhooks about TARGET_ASM_OPEN_PAREN at the end of the build.

Cheers
John

tgingold, 2023-01-10:
Ah sorry, I missed that message.
But it looks like an error independent of ghdl.  I suppose if you try to build gcc without ghdl it will also fail.

OTOH, this is an error while building documentation; so maybe you should try to skip this step.

jrtchapple, 2023-03-08:
Just getting around to revisiting this.

Yes, building gcc without ghdl does fail.

I saw that GHDL is included in bookworm (Debian 12), but it doesn't boot on an Ryzen testbed.  I didn't try it on my main computer, which is Intel based.

I am wondering how to build gcc and ghdl without building the documentation. Searching gcc.pdf on 'build' does not result in any relevant hits.

Cheers
John

tgingold, 2023-03-08:
Maybe try with: make MAKEINFO=true

I remember using this command when doc generation tools were not available (but I am not sure that info is still used).

jrtchapple, 2023-03-24:
Thanks Tristan.

Between using the MAKEINFO trick and GCC 12.2 as a back end, I made it work.  I used the new release (3.0) source code.  The bug I reported last year or so has indeed vanished.
Cheers
John

closed 2023-04-19 by umarcor