ghdl/ghdl #2307: Crash when building Vivado's 2022.1 libraries

Palmitoxico, 2023-01-10:
**Description**
GHDL crashes when building Vivado's 2022.1 `unisim` library, specifically when compiling `unisims/primitive/ISERDES_NODELAY.vhd`. This is a regression since older versions worked fine. I've bisected and found that 5707cc82d8222346b475fb820be81fd677bc3c29 introduced the problem. I can't share the offending file due to Xilinx's copyright.

**How to reproduce?**
Build Vivado libraries:
```
$ /usr/local/lib/ghdl/vendors/compile-xilinx-vivado.sh -a --vhdl2008 --output xilinx-vivado --source /opt/xilinx/Vivado/2022.1/data/vhdl/src/
```

**Context**
- OS: Archlinux
- Origin:
  - [ ] Built from sources: `5707cc82d8222346b475fb820be81fd677bc3c29`

```
internal error: eval_monadic_operator: IIR_PREDEFINED_NONE

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r927.g5707cc82d) [Dunoon edition]
Compiled with GNAT Version: 10.2.1 20210110
Target: x86_64-linux-gnu
/home/augusto.fraga/repos/ghdl/
Command line:
/tmp/ghdl/bin/ghdl1-llvm --mb-comments -fexplicit -Wbinding -Wno-hide -fsynopsys --no-vital-checks --std=08 -frelaxed --work=unisim --workdir=/home/augusto.fraga/xilinx-vivado/unisim/v08 -P/tmp/test -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -fpic -c -o /tmp/test/ISERDES_NODELAY.o /opt/xilinx/Vivado/2022.1/data/vhdl/src/unisims/primitive/ISERDES_NODELAY.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:181
Call stack traceback locations:
0x4faf3d 0x5c1ae8 0x5cbc84 0x5cbe31 0x5cc3ac 0x6654f5 0x657068 0x64c700 0x64c926 0x64cf81 0x656566 0x6570a8 0x64c74f 0x64c926 0x6567b4 0x6570a8 0x64c700 0x64c926 0x64cf81 0x657094 0x64c74f 0x64c926 0x6567b4 0x6570a8 0x658032 0x658bcc 0x6682de 0x66af40 0x669dce 0x66b065 0x66af51 0x66b13f 0x66bd78 0x66be4c 0x66c5dc 0x66c85a 0x66c9c7 0x631ead 0x639463 0x60defc 0x72c2d6 0x72e054 0x72f6a7 0x40caf3 0x7fbc0a96dd08 0x40b778 0xfffffffffffffffe
******************************************************************
```

diogratia, 2023-01-11:
Got essentially the same message with a newer commit and the 14.7/ISE_DS version:

```none
%: ghdl -a --std=08 --work=unisim unisim_VCOMP.vhd
%: ghdl -a --std=08 -frelaxed --work=unisim unisim_VPKG.vhd
unisim_VPKG.vhd:993:9:warning: declaration of "i" hides variable "i" [-Whide]
    for i in 1 to full_nibble_count loop
        ^
%: ghdl -a --std=08 -frelaxed -fsynopsys ISERDES_NODELAY.vhd
ISERDES_NODELAY.vhd:750:26:warning: declaration of "interface_type" hides generic "interface_type" [-Whide]
  procedure CR454107_msg(INTERFACE_TYPE : IN string;
                         ^
ISERDES_NODELAY.vhd:751:26:warning: declaration of "data_rate" hides generic "data_rate" [-Whide]
                         DATA_RATE  : IN string;
                         ^
ISERDES_NODELAY.vhd:752:26:warning: declaration of "data_width" hides generic "data_width" [-Whide]
                         DATA_WIDTH : IN integer) is
                         ^
internal error: eval_monadic_operator: IIR_PREDEFINED_NONE

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r1001.g8845f761e) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Users/dio_gratia/Desktop/opt/Xilinx/14.7/ISE_DS/vhdl/src/unisims/
Command line:
/Users/dio_gratia/opt/share/ghdl/bin/ghdl1-llvm --std=08 -frelaxed -fsynopsys -P/Users/dio_gratia/opt/share/ghdl/lib/ghdl/ieee/v08/ -P/Users/dio_gratia/opt/share/ghdl/lib/ghdl/ -c -o ISERDES_NODELAY.o ISERDES_NODELAY.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:181
Load address: 0x100058000
Call stack traceback locations:
0x100145ef3 0x10020b961 0x10021651d 0x1002166c7 0x100216c34 0x1002b340a 0x1002a31ab 0x100298208 0x100298430 0x100298a9f 0x1002a2284 0x1002a31eb 0x100298257 0x100298430 0x1002a24ac 0x1002a31eb 0x100298208 0x100298430 0x100298a9f 0x1002a31d7 0x100298257 0x100298430 0x1002a24ac 0x1002a31eb 0x1002a41b0 0x1002a4d5e 0x1002b61aa 0x1002b8f0e 0x1002b7d1c 0x1002b9030 0x1002b8f1f 0x1002b9105 0x1002b9f00 0x1002b9fd6 0x1002ba961 0x1002babdc 0x1002bad3b 0x10027c901 0x100284523 0x100256df5 0x100379d45 0x10037b736 0x10037ca14 0x10005b954
******************************************************************
ghdl:error: compilation error
%
```
Not a lot more information analyzing manually mind, Line 752 is in procedure CR454107_msg as an architecture declarative item for architecture ISERDES_NODELAY_V OF ISERDES_NODELAY. That leaves around 1100 lines that could be commented out in a search to localize the problem.

The internal error message is from src/vhdl/vhdl-evaluation.adb and while preexisting shows up in the 5707cc8 change.

By using a sliding search from the end of the architecture I localized an occurrence to process prcs_Q1_rnk1 labelled at line 1087.  By sliding the end of comments up to the end of the process we find there's more than one occurrence that will cause the internal error but it gives us a  set of operations to look at.

It's not perfectly enlightening,  It looks like there might be a missing case in the 5707cc82 change.

```vhdl
  prcs_Q1_rnk1:process(CLK_dly, GSR_dly, REV_dly, SR_dly)
  variable q1rnk1_var         : std_ulogic := TO_X01(INIT_Q1);
  begin
     if(GSR_dly = '1') then
         q1rnk1_var  :=  TO_X01(INIT_Q1);
     elsif(GSR_dly = '0') then
        case AttrSRtype is
           when 0 => 
           --------------- // async SET/RESET
                   if((SR_dly = '1') and (not ((REV_dly = '1') and (TO_X01(SRVAL_Q1) = '1')))) then
                      q1rnk1_var := TO_X01(SRVAL_Q1);
                   elsif(REV_dly = '1') then
                      q1rnk1_var := not TO_X01(SRVAL_Q1);
                   elsif((SR_dly = '0') and (REV_dly = '0')) then
                      if(ice = '1') then
                         if(rising_edge(CLK_dly)) then
                            q1rnk1_var := datain;
                         end if;
                      end if;
                   end if;

           when 1 => 
           --------------- // sync SET/RESET
[... 90 lines cut ...]

```none
%: ghdl -a --std=08 issue_2307_mcve.vhdl
internal error: eval_monadic_operator: IIR_PREDEFINED_NONE

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r1001.g8845f761e) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Users/dio_gratia/Desktop/
Command line:
/Users/dio_gratia/opt/share/ghdl/bin/ghdl1-llvm --std=08 -P/Users/dio_gratia/opt/share/ghdl/lib/ghdl/ieee/v08/ -P/Users/dio_gratia/opt/share/ghdl/lib/ghdl/ -c -o issue_2307_mcve.o issue_2307_mcve.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:181
Load address: 0x1044f1000
Call stack traceback locations:
0x1045deef3 0x1046a4961 0x1046af51d 0x1046af6c7 0x1046afc34 0x10474c40a 0x10473c1ab 0x10473d045 0x10472837f 0x10472beeb 0x10472c23c 0x1047520e5 0x104752f00 0x104752fd6 0x104753961 0x104753bdc 0x104753d3b 0x104715901 0x10471d523 0x1046efdf5 0x104812d45 0x104814736 0x104815a14 0x1044f4954
******************************************************************
ghdl:error: compilation error
%: 
```

It'd be worth checking to see the further errors found in ISERDES_NODELAY.vhd are also corrected.

tgingold, 2023-01-11:
I have a fix for this, testing in progress.

Thank you for the issue!

tgingold, 2023-01-11:
This is due to to_x01, whose result is locally static but not yet handled.

closed 2023-01-11 by tgingold

diogratia, 2023-01-11:
The ISE 14.7 ISERDES_NODELAY.vhd revision 04/07/08 - CR 469973  file successfully analyzes following the Tristan's fix to add support for type BIT when determining whether a function call to TO_X01 is a *locally static primary* (IEEE Std 1076-2008 9.4.2  Locally static primaries) or a *globally static primary* (9.4.3  Globally static primaries).  

Tristan's test case is nice and concise.

Palmitoxico, 2023-01-12:
Thank you @tgingold !

commit 5e13660e (2023-01-11): synth-vhdl_eval: handle to_X01 for bit to std_ulogic.

Fix #2307