ghdl/ghdl #2309: Using GHDL to simulate Xilinx LogiCORE Floating-Point 7.1 IP core

nimou377, 2023-01-11:
Hello,

I have a custom IP that instantiate the Xilinx LogiCORE Floating-Point 7.1 IP.

I can simulate it with Vivado without any issue. But the goal now would be to be able to simulate it using GHDL. 

I have already precompiled the following Xilinx libraries: unisim, unimacro, unifast and secureip. I can simulate with GHDL modules that instantiate primitives with no problem.

But is it possible to simulate an encrypted IP like the LogiCORE Floating-Point 7.1 IP ?

I customized the IP in Vivado, and then generated the output products. The generated IP folder looks like this:

├── cmodel
│   ├── floating_point_v7_1_bitacc_cmodel_lin64.zip
│   └── floating_point_v7_1_bitacc_cmodel_nt64.zip
├── demo_tb
│   └── tb_floating_point_0.vhd
├── doc
│   └── floating_point_v7_1_changelog.txt
├── floating_point_0.dcp
├── floating_point_0_ooc.xdc
├── floating_point_0_sim_netlist.v
├── floating_point_0_sim_netlist.vhdl
├── floating_point_0_stub.v
├── floating_point_0_stub.vhdl
├── floating_point_0.veo
├── floating_point_0.vho
├── floating_point_0.xci
├── floating_point_0.xml
├── hdl
│   ├── axi_utils_v2_0_vh_rfs.vhd
│   ├── floating_point_v7_1_rfs.v
│   ├── floating_point_v7_1_rfs.vhd
│   ├── mult_gen_v12_0_vh_rfs.vhd
│   ├── xbip_bram18k_v3_0_vh_rfs.vhd
│   ├── xbip_dsp48_addsub_v3_0_vh_rfs.vhd
│   ├── xbip_dsp48_multadd_v3_0_vh_rfs.vhd
│   ├── xbip_dsp48_wrapper_v3_0_vh_rfs.vhd
│   ├── xbip_pipe_v3_0_vh_rfs.vhd
│   └── xbip_utils_v3_0_vh_rfs.vhd
├── sim
│   └── floating_point_0.v
└── synth
    └── floating_point_0.v

Folder zip file: [floating_point_0.zip](https://github.com/ghdl/ghdl/files/10393739/floating_point_0.zip)

The only relevant posts I could find searching the internet were these 2:

https://stackoverflow.com/questions/41469436/how-to-use-ghdl-to-simulate-generated-xilinx-ip
https://www.reddit.com/r/FPGA/comments/mpy1d2/any_ghdl_workaround_for_encrypted_ip_cores_from/

Some say that the floating_point_0_sim_netlist.vhdl file should provide a functional netlist that could be used for simulation, but it seems that this file is encrypted itself.

Another suggestion was to generate the netlist using Vivado, and then use it with GHDL. Apart from generating the output products as described above, I'm not sure what I could do regarding this.

Does anyone has any idea on how I could simulate the LogiCORE Floating-Point 7.1 IP with GHDL ? Is it possible ?

Any pointer would be very appreciated.

Thanks !
Ril

tgingold, 2023-01-11:
Unfortunately there is no magic solution.
Encrypted IPs cannot be handled without the key...

Any suggestion is welcome!

nimou377, 2023-01-12:
Hi tgingold, thanks for clearing that out.

BTW, I also ask the question on the Xilinx forums, and it has also been confirmed by a Xilinx employee : https://support.xilinx.com/s/question/0D54U00006CnchZSAR/how-to-simulate-xilinx-logicore-floatingpoint-71-ip-core-using-ghdl-?language=en_US

I hope this will be possible one day!

closed 2023-01-12 by nimou377