ghdl/ghdl #2311: VPI: name of returned toplevel is lowercase

Scafir, 2023-01-11:
**Description**
Using cocotb, one must use lowercase to specify the toplevel name. Otherwise, the comparison fails.
If GHDL cannot provide a correctly cased toplevel name, then this issue could be shadowed on cocotb's side, as the comparison is made [there](https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/vpi/VpiImpl.cpp#L506).

**Expected behaviour**
The correct case should be returned.

**How to reproduce?**
[cocotb_toplevel_case.zip](https://github.com/ghdl/ghdl/files/10394609/cocotb_toplevel_case.zip)

```vhd
-- lowerUPPER.vhd
LIBRARY IEEE;
USE IEEE.std_logic_1164.ALL;
ENTITY lowerUPPER IS
    PORT(
        a : in std_logic;
        b : out std_logic
    );
END lowerUPPER;

ARCHITECTURE structure OF lowerUPPER IS
BEGIN
    b <= a;
END structure;
```

```py
# test_lowerUPPER.py
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
async def INIT_test(dut):
    print("This test is not important")
```

```Makefile
# Makefile
VHDL_SOURCES = $(PWD)/lowerUPPER.vhd
MODULE = test_lowerUPPER
TOPLEVEL = lowerUPPER
SIM = ghdl

include $(shell cocotb-config --makefiles)/Makefile.sim
```

**Context**
Please, provide the following information:

- OS: Centos Stream 8
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `5b458e`

Paebbels, 2023-01-11:
VPI is a Verilog Programming Interface and Verilog is case sensitive. VPI is used by GHDL, because the interface is more lightweight compared to VHPI.

I didn't fully get your point. Is GHDL returning the toplevel in lowercase or is it about cocotb needing to provide the toplevel in lowercase?

Paebbels, 2023-01-11:
So GHDL normalizes the name and it's needed as not normalized. Got it.