ghdl/ghdl #2313: Build libghdl for MinGW64, UCRT64 and Linux with dependencies included

Paebbels, 2023-01-13:
**Is your feature request related to a problem? Please describe.**

I would like to ship precompiled binaries of `libghdl` with Python wheels to ease installation of `pyGHDL` as a standalone package via PIP and PyPI without the need of a full/local GHDL installation.

**Describe the solution you'd like**

`mcode` backend variants are preferred, because:
* no `*.o` files are generated by `libghdl`.
* `mcode` is faster in RAM for small files. Moreover, currently the performance of `llvm` is not needed.
* `libghdl` is currently used for parsing, no synthesis, no simulation.
* as local CPython installations will be 64-bit installations, `mcode 32-bit` is not needed.
* `mcode` has less depdendencies (`*.so`/`*.dll` files)

When looking at MinGW64 and UCRT64 variants of GHDL, these dependencies are used:
* `libgcc_s_seh-1.dll`
* `libwinpthread-1.dll`
* `libgnat-12.dll`

It looks like all of them have an equivalent `*.a` file, so the linker could link all dependencies together into a single `libghdl` shared library.

GHDL's libraries `libghdlvpi` and `libghw` are currently not needed for `pyGHDL`.

**Describe alternatives you've considered**

The current alternative is building and installing GHDL from sources using MSYS2/MinGW64. Afterwards, some DLLs are copied manually into the GHDL installation directory to create a standalone variant.

Such a standalone variant is also provided via nightly builds as a ZIP file. Unfortunately, updating requires again manual downloads and overwriting files.

**Additional context**

Python's wheel packages can:
* contain platform independent Python code
* platform specific precompiled binaries
* source code, that's compiled on the target machine using locally available compilers  
  (not every machine has an Ada compiler - and Windows even lacks a C compiler)

Paebbels, 2023-01-13:
I think only a variant for `pyGHDL` should use such a shared library. It's not intended for GHDL to use such a all-in-one shared library.

A proof of concept can be started for the MSYS2 environments MinGW64 and UCRT64. When this works, it can be expanded to Linux and macOS.

chenailuoding1, 2024-11-24:
I installed GHDL on Windows through MSYS2. Then I wanted to use pyGHDL in Python, but an error occurred.May I ask you some questions

Paebbels, 2024-11-24:
Please see my answer in #2808.

closed 2024-11-24 by Paebbels