ghdl/ghdl #2316: Uninstall leaves files on disk

Paebbels, 2023-01-13:
**Description**

When calling `make uninstall`, some files remain on disk.

* `include/ghdl/synth.h`
* `include/ghdl/synth_gates.h`
* `libghdl.a`
* `libghdl.link`
* `libghw.dll`

**Expected behaviour**

As many files of GHDL are removed from disk, I suspect these files should get deleted too. Especially as these files are "new" to GHDL ;). I assume an oversight when adding rules to the `Makefile.in`.

**How to reproduce?**

Tested with MSYS2/MinGW32 mcode.

```
mkdir -p build/ming32-mcode
cd build/ming32-mcode
../../configure --prefix=/c/Tools/ghdl/3.0.0.dev0-mingw32-mcode
make
make install
make uninstall
```

**Additional context**

Output of `PS> dir | tree /f`
```
├───bin
├───include
│   └───ghdl
│           synth.h
│           synth_gates.h
│
└───lib
        libghdl.a
        libghdl.link
        libghw.dll
```

OmarElfouly, 2024-10-13:
Hello @Paebbels 

I'm new to contributing and I'm excited to get started on this issue :)

I just wanted to confirm that the expected behavior is to remove the files that were listed and that the recommended starting point is MakeFile.in

Paebbels, 2024-10-13:
I'm currently reworking the GHDL pipeline, maybe such a check should be part of the pipeline.

OmarElfouly, 2024-10-13:
Sounds like a great idea.
Also I'll make a start on going through the MakeFile :)

OmarElfouly, 2024-10-19:
I panicked for a bit when i saw the build check fail but it looks like that's an issue that exists on master. Is that right @Paebbels

Paebbels, 2024-10-20:
To which pipeline are you referring? Can you post a pipeline link?

I activated the pipeline for your PR. It's currently running.

PR #2783 (2024-10-22): Modifies uninstall for Uninstall leaves files on disk #2316
**Description**

Modifies Makefile.in to properly remove all files. Modification was added to uninstall.libghdl as it is what I currently beelive is the most suitable location.

This allows for a complete uninstall without leaving any files behind to close #2316
closed 2024-10-22 by tgingold