ghdl/ghdl #2317: 'make install' doesn't work without standalone 'make'

Paebbels, 2023-01-13:
**Description**

When calling `./configure` followed by `make`, it fails due to missing dependencies. It looks like VHDL packages from `lib/ghdl/ieee/v87/*` are not ready to be installed.

**Expected behaviour**

In Linux like environments, `make` is executed in two steps, so building software run with user privileges and installation runs with root privileges (if needed).

```bash
make
sudo make install
```

In an MSYS2 environment this distinguishing is now needed, thus `make install` should provide same behavior as calling in 2 steps.

**How to reproduce?**

```bash
mkdir -p build/ming32-mcode
cd build/ming32-mcode
../../configure --prefix=/c/Tools/ghdl/3.0.0.dev0-mingw32-mcode
make install
```

I assume the same behavior for MinGW32/MinGW64 and UCRT64.

**Additional context**

```
install: der Aufruf von stat für 'lib/ghdl/ieee/v87/*' ist nicht möglich: No such file or directory
install: der Aufruf von stat für 'lib/ghdl/ieee/v93/*' ist nicht möglich: No such file or directory
install: der Aufruf von stat für 'lib/ghdl/ieee/v08/*' ist nicht möglich: No such file or directory
install: der Aufruf von stat für 'lib/ghdl/std/v87/*' ist nicht möglich: No such file or directory
install: der Aufruf von stat für 'lib/ghdl/std/v93/*' ist nicht möglich: No such file or directory
install: der Aufruf von stat für 'lib/ghdl/std/v08/*' ist nicht möglich: No such file or directory
install: Verzeichnis 'lib/ghdl/src/ieee/v87' wird ausgelassen
install: Verzeichnis 'lib/ghdl/src/ieee/v93' wird ausgelassen
install: der Aufruf von stat für 'lib/ghdl/src/ieee/v87/*' ist nicht möglich: No such file or directory
install: der Aufruf von stat für 'lib/ghdl/src/ieee/v93/*' ist nicht möglich: No such file or directory
install: der Aufruf von stat für 'lib/ghdl/src/ieee2008/*' ist nicht möglich: No such file or directory
install: Verzeichnis 'lib/ghdl/src/std/v08' wird ausgelassen
install: Verzeichnis 'lib/ghdl/src/std/v87' wird ausgelassen
install: Verzeichnis 'lib/ghdl/src/std/v93' wird ausgelassen
install: der Aufruf von stat für 'lib/ghdl/src/std/v87/*' ist nicht möglich: No such file or directory
install: der Aufruf von stat für 'lib/ghdl/src/std/v93/*' ist nicht möglich: No such file or directory
install: der Aufruf von stat für 'lib/ghdl/src/std/v08/*' ist nicht möglich: No such file or directory
install: Verzeichnis 'lib/ghdl/src/synopsys/v08' wird ausgelassen
install: der Aufruf von stat für 'lib/ghdl/src/synopsys/v08/*' ist nicht möglich: No such file or directory
install: der Aufruf von stat für 'lib/ghdl/src/upf/*' ist nicht möglich: No such file or directory
install: der Aufruf von stat für 'lib/ghdl/src/vital95/*' ist nicht möglich: No such file or directory
install: der Aufruf von stat für 'lib/ghdl/src/vital2000/*' ist nicht möglich: No such file or directory
make: *** [Makefile:653: install.vhdllib] Fehler 1
```

tgingold, 2023-01-13:
Ok, but before 'make install' you need to do 'make'.

This is the procedure to build ghdl.

Paebbels, 2023-01-13:
So `make install` without `make` is no usecase for GHDL?

Then we can close this as rejected, right?

tgingold, 2023-01-13:
I would say so.  It's a very minor issue.

closed 2023-01-13 by Paebbels