ghdl/ghdl #2319: Unexpected run termination

dinvert, 2023-01-14:
I'm seeing an odd behaviour when running a testbench design for a small processor prototype. Depending on the sequence of instructions run on the simulated core, the GHDL run sometimes terminates early and unexpectedly. So for example this is the expected output:

```
ghdl -e compact_tb
ghdl -r compact_tb --stop-time=5us --wave=compact_tb.ghw
wait...
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@20ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@20ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@20ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
./compact_tb:info: simulation stopped by --stop-time @5us

```
(assertion warnings are expected).

Now with the same design but different instructions run on the core, the output is 

```
ghdl -e compact_tb
ghdl -r compact_tb --stop-time=5us --wave=compact_tb.ghw
wait...
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@20ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@20ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@20ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
./compact_tb:info: simulation stopped @420ns by **--stop-delta=5000**
```

There is no stop delta on the command line. I'm assuming 5000 is a default value. If a stop delta value is actually passed on the command line, it appears in the output, but the run still stops at 420ns, regardless of the given value.

If this is really unexpected behaviour, what is the best way to analyze this further and pinpoint the particular event that triggers a stop at that particular simulated time? I tried process tracing, but didn't see anything immediately obvious in the output.

Or is this potentially a plausible scenario and I'm just mislead by the output? Perhaps there is some default steady state detection?

GHDL 3.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 6.5.0
 llvm code generator
(Linux, built from current master; although also seen in older versions)

dinvert, 2023-01-14:
Digging a bit more and passing --disp-time, it simply seems that no new signal events are seen after that particular time and the run terminates (correctly?) after 420ns+5000. I'm not sure how that is possible though, at the very least the clock should still be toggling, even if everything else were in a steady state?

tgingold, 2023-01-14:
Please provide a reproducer or a testcase.

Your design is probably not making any time progress and you hit the delta time limit.

dinvert, 2023-01-14:
Ok I'll try to come up with something to reproduce the problem.
That's just what puzzles me. Time progress is exactly as expected if I take the CPU component out altogether so that only the clock is still running. Almost as if one stalled process (cpu in halt) pauses the entire design.

dinvert, 2023-01-14:
Found the issue, XSim gave some clues - there's a zero delay oscillation somewhere within the CPU. Definitely not a GHDL issue. Please close- not a GHDL specific problem at all. Thanks for the input!

closed 2023-01-14 by dinvert