ghdl/ghdl #2323: Range constraint of std_logic_vector is not respected

jwprice100, 2023-01-19:
**Description**
The natural range constraint of std_logic_vector (or std_ulogic_vector) does not appear to be respected.  It will compile even with a negative bound.

**Expected behaviour**
A range constraint with a negative bound should result in a compile error.

**How to reproduce?**
Have a signal of type std_logic_vector with a negative bound. GHDL will compile it despite the range being natural. 

```vhd :file: test_ip.vhd
library ieee;
use ieee.std_logic_1164.all;

entity TEST_IP is
port (
  I_CLK                   : in std_logic;
  I_RST                   : in std_logic;

  I_DATA                  : in std_logic;
  O_DATA                  : out std_logic
);
end entity TEST_IP;

architecture TEST_IP_ARCH of TEST_IP is

  signal test_signal      : std_logic_vector(7 downto -1);

begin

end architecture TEST_IP_ARCH;
```
**Context**
Please, provide the following information:
- OS: Ubuntu 22.04
- Origin:
  - [X] Built from sources: `27812b602d2dfc452895c156bf0376bdcda71c8c`

tgingold, 2023-01-22:
You get a warning during analysis, but you should indeed get an error during simulation.

diogratia, 2023-01-22:
An error occurs during elaboration:

IEEE Std 1076-1993 (you get the same result in later revisions)

12.3.1.4 Object declarations  
>  
>Elaboration of an object declaration that declares an object other than a file object proceeds as follows:  
>  
>>a. The subtype indication is first elaborated. This establishes the subtype of the object.
>>...  

12.3.1.3 Subtype declarations  
>  
>Elaboration of a subtype declaration consists of the elaboration of the subtype indication. The elaboration of a subtype indication creates a subtype. If the subtype does not include a constraint, then the subtype is the same as that denoted by the type mark. The elaboration of a subtype indication that includes a constraint proceeds as follows:  
>  
>>a. The constraint is first elaborated.  
>>b. A check is then made that the constraint is compatible with the type or subtype denoted by the type mark (see 3.1  and 3.2.1.1).   
>  
>Elaboration of a range constraint consists of the evaluation of the range. The evaluation of a range defines the bounds and direction of the range. Elaboration of an index constraint consists of the elaboration of each of the discrete ranges in the index constraint in some order that is not defined by the language.

3.2.1.1 Array constraints

An index constraint is *compatible* with the type denoted by the type mark if and only if the constraint defined by each discrete range is compatible with the corresponding index subtype. If any of the discrete ranges defines a null range, any array thus constrained is a *null array*, having no components. An array value *satisfies* an index constraint if at each index position the array value and the index constraint have the same index range. (Note, however, that assignment and certain other operations on arrays involve an implicit subtype conversion.) 

4.2 Subtype declarations

>A subtype indication defines a subtype of the base type of the type mark.
>    
>...  
>  
>If the subtype indication does not include a constraint, the subtype is the same as that denoted by the type mark. The condition imposed by a constraint is the condition obtained after evaluation of the expressions and ranges forming the constraint. The rules defining compatibility are given for each form of constraint in the appropriate section. These rules are such that if a constraint is compatible with a subtype, then the condition imposed by the constraint cannot contradict any condition already imposed by the subtype on its values. An error occurs if any check of compatibility fails. 

Here the check of compatibility occurs during elaboration of the subtype indication as described in 12.3.1.3 b. The index constraint of the test_signal object declaration is not compatible with the type mark std_logic_vector which has a natural  index range (see the declaration for IEEE package std_logic_1164).

A quick check shows the compatibility isn't checked for a constraint on a scalar type mark either.

commit c92813bb (2023-01-26): vhdl: improve range checks, fix #2323
closed 2023-01-26 by tgingold