ghdl/ghdl #2324: Record support in VPI

Scafir, 2023-01-20:
**Description**
It is currently not possible to access records using the VPI. This is a known issue, but currently does not have a bug report associated to it. I hope making this one will help tracking and awareness of the issue.

Related materials:
* Merge request (WIP): #1249
* Cocotb related issue: https://github.com/cocotb/cocotb/issues/2591
* Related issue: #237 

**Expected behaviour**
Records should be accessible through the VPI.

**How to reproduce?**
[records.zip](https://github.com/ghdl/ghdl/files/10466793/records.zip)

```vhd
-- sample.vhd
library ieee;
use ieee.std_logic_1164.all;

package recordPkg is

    type myRecordT is record
        a : std_logic;
    end record myRecordT;

end package recordPkg;

library ieee;
use ieee.std_logic_1164.all;
use work.recordPkg.all;

entity sample is
    port(
        rec : in myRecordT
    );
end sample;

architecture structure of sample is
begin
end structure;
```

```py
# test_sample.py
import cocotb
from cocotb.triggers import RisingEdge, ClockCycles, Timer, FallingEdge
from cocotb.clock import Clock

@cocotb.test()
async def type_test(dut):
    # Reading
    print(dut.rec.a)

    # Writing
    dut.rec.a = 0

    await Timer(1, 'step')

    assert dut.rec.a == 0, "rec.a was not updated"
```

```Makefile
# Makefile
VHDL_SOURCES = $(PWD)/sample.vhd
MODULE = test_sample
TOPLEVEL = sample
SIM = ghdl
SIM_ARGS = "--wave=a.ghw"

include $(shell cocotb-config --makefiles)/Makefile.sim
```

**Context**
Please, provide the following information:

- OS: Fedora Linux 37 (Workstation Edition)
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 65a40a067d1b153265eaf8bcadbc225c5eeac041

tgingold, 2024-12-21:
Yes, this has to be implemented.

AstridWinkler, 2025-09-05:
The workaround I use is a top module to access the signals from the record (not that practical at all even more so for complex multi-source architecture).
What other solution have you found ?
Here is a working sample

```{r}
--- module.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

package package_module is
    type T_module_i is
        record
            clk       : std_logic;
            reset     : std_logic;
            i_signal  : std_logic_vector(16-1 downto 0);
        end record;

    type T_module_o is
        record
            o_signal  : std_logic_vector(16-1 downto 0);
        end record;
end package_module;

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use work.package_module.all;

entity module is
    port (
        i : in  T_module_i;
        o : out T_module_o
    );
end module;

architecture arch of module is
begin
    process(i.clk, i.reset)
    begin
        if i.reset = '1' then
            o.o_signal <= (others=>'0');
        else rising_edge(i.clk) then
            o.o_signal <= std_logic_vector(unsigned(i.i_signal + 1));
        end if;
    end;
end arch ;
```

```{r}
-- top_module.vhd (mapping)
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use work.package_module.all;

entity module_top is
    port (
        clk         : in  std_logic;
        reset       : in  std_logic;
        i_signal    : in  std_logic_vector(16-1 downto 0);
        o_signal    : out std_logic_vector(16-1 downto 0);
    );
end entity;

architecture rtl of module_top is
    signal i_rec : T_module_i;
    signal o_rec : T_module_o;
begin

    i_rec.clk      <= clk;
    i_rec.reset    <= reset;
    i_rec.i_signal <= i_signal;
    o_signal       <= o_rec.o_signal;

    UUT: entity work.module
        port map (
            i => i_rec,
            o => o_rec
        );
end architecture;
```

```{r}
# test_module.py
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge

@cocotb.test()
async def test(dut):

    cocotb.start_soon(Clock(dut.clk, 200, units="ns").start())

    for i in range(10):
        dut.i_signal.value = i
        await RisingEdge(dut.clk)
    
    assert dut.o_signal.value == dut.o_signal.value, "output != input"
```

tgingold, 2025-09-09:
@AstridWinkler It's on the schedule...

anirudh12032008, 2026-11-06:
I'd like to work on this, unless you already have it in progress @tgingold. #1249 is very old so I'd start fresh. Right now records are dropped in vpi_scan because Vhpi_Handle_To_Vpi_Net goes through the VCD wire info, which has no record kind (Vcd_Bad). Plan: give record signals/ports a struct handle (vpiStructNet), let vpi_iterate(vpiMember, h) walk the elements using the record RTI and element offsets, so nested records and arrays inside records work the same way, with get/put value only on the leaf members. Does that match what you had in mind?

tgingold, 2026-11-07:
@anirudh12032008 feel free to work on it.

My larger plan is to rework the vhpi/vpi interface and to base it on `simul-vhdl_elab` constructs to get rid of the limitations of `grt-rti`.

closed 2026-11-08 by tgingold