ghdl/ghdl #2325: building tag 2.0.0

joealsubash, 2023-01-21:
Hi, thanks for this great tool but have run into some issues.

Trying to build and install ghdl locally to use ghdl-ls with emacs on  a fedora 36 machine. Don't have admin access on the machine so after installing gnat at ```/home/user/opt/GNAT``` I download https://github.com/ghdl/ghdl/releases/tag/v2.0.0

after unzipping in folrder with user wrx permissions I run

```sh configure --prefix=/home/user/opt/ghdl-2.0.0 --with-gcc=/home/user/opt/GNAT/2021/bin```
followed by
```make```
followed by 
```make install```

which gives me this error

```
  install -m 644 /home/user/opt/ghdl-2.0.0/lib/ghdl/libbacktrace.a /home/user/opt/ghdl-2.0.0/lib/ghdl/libbacktrace.a
for d in ieee/v87 ieee/v93 ieee/v08 std/v87 std/v93 std/v08 src/ieee src/ieee/v87 src/ieee/v93 src/ieee2008 src/std src/std/v87 src/std/v93 src/std/v08 src/synopsys src/synopsys/v08 src/upf; do \
  mkdir -p /home/user/opt/ghdl-2.0.0/lib/ghdl/$d; \
  install -m 644 -p \
    lib/ghdl/$d/* /home/user/opt/ghdl-2.0.0/lib/ghdl/$d; \
done
install: cannot stat 'lib/ghdl/ieee/v87/*': No such file or directory
install: cannot stat 'lib/ghdl/ieee/v93/*': No such file or directory
install: cannot stat 'lib/ghdl/ieee/v08/*': No such file or directory
install: cannot stat 'lib/ghdl/std/v87/*': No such file or directory
install: cannot stat 'lib/ghdl/std/v93/*': No such file or directory
install: cannot stat 'lib/ghdl/std/v08/*': No such file or directory
install: omitting directory 'lib/ghdl/src/ieee/v87'
install: omitting directory 'lib/ghdl/src/ieee/v93'
install: cannot stat 'lib/ghdl/src/ieee/v87/*': No such file or directory
install: cannot stat 'lib/ghdl/src/ieee/v93/*': No such file or directory
install: cannot stat 'lib/ghdl/src/ieee2008/*': No such file or directory
install: omitting directory 'lib/ghdl/src/std/v08'
install: omitting directory 'lib/ghdl/src/std/v87'
install: omitting directory 'lib/ghdl/src/std/v93'
install: cannot stat 'lib/ghdl/src/std/v87/*': No such file or directory
install: cannot stat 'lib/ghdl/src/std/v93/*': No such file or directory
install: cannot stat 'lib/ghdl/src/std/v08/*': No such file or directory
install: omitting directory 'lib/ghdl/src/synopsys/v08'
install: cannot stat 'lib/ghdl/src/synopsys/v08/*': No such file or directory
install: cannot stat 'lib/ghdl/src/upf/*': No such file or directory
make: *** [Makefile:602: install.vhdllib] Error 1
```
the ```lib``` folder created in the directory where I run make is has V87, v93 etc folders which are empty. libghdl-2.0.0.so and libghw.so are however created. 
This looks like an issue with the build process.

tgingold, 2023-01-21:
I think you should build like this:

PATH=/home/user/opt/GNAT/2021/bin:$PATH
sh configure --prefix=/home/user/opt/ghdl-2.0.0
make
make install

joealsubash, 2023-01-21:
Thanks for your reply.
Tried your before using the ```--with-gcc``` flag. With GNAT in ```PATH``` and without the ```--with-gcc``` flag end up with linker errors.

```
gcc -c -I./ -gnat12 -I./src -I./src/vhdl -I./src/synth -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -I./src/synth -gnat12 -gnaty3befhkmr -g -gnatwe -gnatwa -gnatwC -gnatf -gnata -gnatw.A
 -I- /disk/homedisk/home/user/ghdl-ls/ghdl-2.0.0/src/synth/synth-ieee-std_logic_1164.adb
gnatbind -aI./src -aI./src/vhdl -aI./src/synth -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/mcode -aI./src/synth -E -x ghdl_jit.ali
gnatlink ghdl_jit.ali -o ghdl_mcode -g memsegs_c.o chkstk.o jumps.o times.o grt-cstdio.o grt-cgnatrts.o grt-cvpi.o grt-cvhpi.o grt-cdynload.o fstapi.o lz4.o fastlz.o -ldl -lm -Wl,--version-script=/home/user/ghdl-ls/ghdl-2.0.0/./src/grt
/grt.ver -Wl,--export-dynamic
/disk/homedisk/home/user/opt/GNAT/2021/bin/../libexec/gcc/x86_64-pc-linux-gnu/10.3.1/ld: .gnu.build.attributes has both ordered [`.gnu.build.attributes.lo.exit' in /lib/../lib64/crt1.o] and unordered [`.gnu.build.attributes' in /lib/..
/lib64/crt1.o] sections
/disk/homedisk/home/user/opt/GNAT/2021/bin/../libexec/gcc/x86_64-pc-linux-gnu/10.3.1/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
gnatlink: error when calling /disk/homedisk/home/user/opt/GNAT/2021/bin/gcc
gnatmake: *** link failed.
make: *** [Makefile:204: ghdl_mcode] Error 4
```

tgingold, 2023-01-22:
What is your distribution ?  I would use gcc-ada from your distribution instead of the compiler you downloaded.

Paebbels, 2024-12-19:
@tgingold any actions on this one or is it outdated?

tgingold, 2024-12-22:
Yes, outdated.

closed 2024-12-22 by tgingold