ghdl/ghdl #2328: Support for VHDL-2008 unary operators on bits

FeepingCreature, 2023-01-28:
**Description**
Follow-up to https://github.com/ghdl/ghdl/issues/1342 : GHDL does not support unary ops on bit_vector.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: test.vhd
entity test is
end test;

architecture behavior of test is
signal a: bit_vector(3 downto 0);
signal b: bit;
begin
	b <= and a;
end behavior;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl --synth --std=08 test.vhd -e test
```

tgingold, 2023-01-29:
I am testing a fix for this.

I haven't seen much use of bit (instead of std_logic) in designs.  Are you also using numeric_bit ?

closed 2023-01-29 by tgingold

FeepingCreature, 2023-01-29:
Nope! I'm a newcomer, and I haven't seen any compelling reason to need the extra states of std_logic.

tgingold, 2023-01-29:
Bit is fine, but std_logic/std_logic_vector is much more standard.  If you want to connect with other cores, it will be simpler.

Also std_logic is initialized by default to 'U', so it's easier to detect reset issues.

commit e6ebb298 (2023-01-29): synth: handle bit reduction operators.  Fix #2328