ghdl/ghdl #2329: Duplicated signal assignment when exporting to verilog

jotego, 2023-01-28:
**Description**

When converting a VHDL design to verilog, a signal connection is not well parsed.

The signal alu_q is connected to a module output (line 5845) but it's also assigned a value (line 5649). The value that it gets assigned is itself. The lines looks like:

```
  assign alu_q = alu_n130; // (signal)
  assign alu_n130 = alu_q; // (signal)

  t65_alu alu (
    .mode(mode_r),
    .op(alu_op_r),
    .busa(busa_r),
    .busb(busb),
    .p_in(p),
    .p_out(alu_p_out),
    .q(alu_q));
```

alu_q also appears declated twice.

**How to reproduce?**
I have attached the required files in [t65.zip](https://github.com/ghdl/ghdl/files/10527979/t65.zip). Unzip and run the following commands.

```
ghdl -a -fsynopsys @t65.f
ghdl synth --out=verilog T65 > t65.v
```

Note that the generated verilog will also contain the *break* keyword used as a variable name, because the original VHDL code did that. This issue report is not about that, although it isn't nice that GHDL doesn't pick up on the keyword use.

**Context**
Please, provide the following information:

- OS: Ubuntu
- Origin: GHDL 3.0.0-dev (v2.0.0-1267-gd82550423) [Dunoon edition]

tgingold, 2023-01-29:
I am investigating the loop assignment.

For `break`, I think this is only a systemverilog keyword.  There is already a mechanism to avoid using verilog keywords, do you want to extend to sv ?

jotego, 2023-01-29:
Thank you.

You're right about the `break` keyword being System Verilog. I guess I had System Verilog support enabled on Verilator when I linted **ghdl** output.

It would be easier if System Verilog keywords were not used either but I understand that it shouldn't be needed (ideally).

tgingold, 2023-01-29:
This is an aliasing in the output.

In T65.vhd line 221, as a workaround rename `alu : T65_ALU` to `inst_alu : T65_ALU`.

jotego, 2023-11-16:
A similar problem occurs when trying to convert the T80s defined here [t80.zip](https://github.com/ghdl/ghdl/files/13372623/t80.zip)

Indeed, by renaming instance names and some ports, the problem goes away. It would be nice if the tool could handle these issues automatically. But, at least there is a manual work around.

I'm closing the issue as it looks like no action has been taken on the code development side and it probably is adding "noise" to the issue list.

Thanks

tgingold, 2023-11-16:
I still plan to work on that...  I prefer to keep it open.