ghdl/ghdl #2330: Regression: raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : elab-vhdl_annotations.adb:1429

donnie-j, 2023-01-29:
**Description**
When running our standard test iCE40 flow for J-Core J1, GHDL now reports an internal error.

**Expected behaviour**
GHDL should, though the yosys plugin, synthesize the J1 iCE40 design correctly.

**How to reproduce?**
Clone https://github.com/j-core/jcore-j1-ghdl.git

```
$ make cpu_lattice_tb
...
$ ghdl --synth cpu_lattice
lattice_spr_wrap.vhd:54:3:warning: instance "r0" of component "sb_spram256ka" is not bound [-Wbinding]
  r0 : SB_SPRAM256KA port map (clock => clk, address => addr(13 downto 0), datain => di(31 downto 16),
  ^
lattice_spr_wrap.vhd:24:14:warning: (in default configuration of bulk_ram(behavioral))
lattice_spr_wrap.vhd:59:3:warning: instance "r1" of component "sb_spram256ka" is not bound [-Wbinding]
  r1 : SB_SPRAM256KA port map (clock => clk, address => addr(13 downto 0), datain => di(15 downto  0),
  ^
lattice_spr_wrap.vhd:24:14:warning: (in default configuration of bulk_ram(behavioral))
lattice_spr_wrap.vhd:64:3:warning: instance "r2" of component "sb_spram256ka" is not bound [-Wbinding]
  r2 : SB_SPRAM256KA port map (clock => clk, address => addr(13 downto 0), datain => di(31 downto 16),
  ^
lattice_spr_wrap.vhd:24:14:warning: (in default configuration of bulk_ram(behavioral))
lattice_spr_wrap.vhd:69:3:warning: instance "r3" of component "sb_spram256ka" is not bound [-Wbinding]
  r3 : SB_SPRAM256KA port map (clock => clk, address => addr(13 downto 0), datain => di(15 downto  0),
  ^
lattice_spr_wrap.vhd:24:14:warning: (in default configuration of bulk_ram(behavioral))

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r1301.g8cb293a92) [Dunoon edition]
Compiled with GNAT Version: 10.2.1 20210110
Target: x86_64-linux-gnu
/home/jeff/work/old/foo/
Command line:
ghdl --synth cpu_lattice
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : elab-vhdl_annotations.adb:1429
Call stack traceback locations:
0x7f825caa17c2 0x5609f494fdea 0x5609f494b4e3 0x5609f494ba7d 0x5609f494bc32 0x5609f494c005 0x5609f494cb08 0x5609f494cc4e 0x5609f494c647 0x5609f494e626 0x5609f4a07d9e 0x5609f4b0bcea 0x5609f4ac8186 0x5609f4b14d58 0x5609f485643b 0x7f825c671d08 0x5609f4854fb8 0xfffffffffffffffe
******************************************************************
```

**Context**
The tools were build with  https://github.com/j-core/openlane-vhdl-build script 02 on Linux.  This means that the LLVM backend is in use.

donnie-j, 2023-01-29:
It appears the regression appears between hashes `1b47cbc9dc619a0788d15954cea61f8d79f4d4fb` and `39a250b87928a2bd0f8da73a2b344ff2688515d2`

Apologies, we haven't run tools regression tests in a few weeks, it seems.

tgingold, 2023-01-29:
It's fine.  Now fixed.

Thank you for the issue.

closed 2023-01-29 by tgingold

donnie-j, 2023-01-29:
Confirm fixed, with thanks :)

commit 3965e755 (2023-01-29): synth: avoid a crash on subtype indication.  Fix #2330