ghdl/ghdl #2331: Xilinx unisim not working:  unit "vital_timing" not found

catkira, 2023-01-30:
I am using the latest GHDL from github.

I built unisim with 
```
compile-xilinx-vivado.sh --unisim --source /tools/Xilinx/Vivado/2022.2/data/vhdl/src
```
then I ran
```
 ghdl -m -g -Pxilinx-vivado --work=int_fft_single_path -fsynopsys int_fft_single_path
```
and get
```
/tools/Xilinx/Vivado/2022.2/data/vhdl/src/unisims/unisim_VPKG.vhd:48:10:error: unit "vital_timing" not found in library "ieee"
/tools/Xilinx/Vivado/2022.2/data/vhdl/src/unisims/unisim_VPKG.vhd:49:10:error: unit "vital_primitives" not found in library "ieee"
/tools/Xilinx/Vivado/2022.2/data/vhdl/src/unisims/unisim_VPKG.vhd:66:16:error: no declaration for "vitaltablesymboltype"
/tools/Xilinx/Vivado/2022.2/data/vhdl/src/unisims/unisim_VPKG.vhd:67:16:error: no declaration for "vitaltablesymboltype"
/tools/Xilinx/Vivado/2022.2/data/vhdl/src/unisims/unisim_VPKG.vhd:68:16:error: no declaration for "vitaltablesymboltype"
/tools/Xilinx/Vivado/2022.2/data/vhdl/src/unisims/unisim_VPKG.vhd:69:16:error: no declaration for "vitaltablesymboltype"
```

does anybody have an idea what went wrong?

tgingold, 2023-01-31:
There is something weird, because during analysis the vital packages were found.

Can you do:
  ghdl --dir ieee

catkira, 2023-01-31:
```
benja@DESKTOP:/mnt/d/git$ ghdl --dir ieee
# Library ieee
# Directory: /usr/local/lib/ghdl/ieee/v93/
package body std_logic_misc
package std_logic_signed
package body std_logic_signed
package std_logic_arith
package body std_logic_arith
package vital_memory
package vital_primitives
package vital_timing
package upf
package math_complex
package math_real
package numeric_std
package numeric_bit
package std_logic_1164
package body std_logic_1164
package body numeric_bit
package body numeric_std
package body math_real
package body math_complex
package body upf
package body vital_timing
package body vital_primitives
package body vital_memory
package std_logic_unsigned
package body std_logic_unsigned
package std_logic_misc
package std_logic_textio
package body std_logic_textio
```
My workaround at the moment is:
```ghdl -a --work=unisim --ieee=synopsys /tools/Xilinx/Vivado/2022.2/data/vhdl/src/unisims/*.vhd /tools/Xilinx/Vivado/2022.2/data/vhdl/src/unisims/primitive/*.vhd```
and then just `-Punisim` when I use GHDL.

Paebbels, 2023-01-31:
Is this a v93 vs. v08 problem?

You can run the `compile-***` scripts with -v for verbose or even -d for debug and see how the files are analyzed and what options have been used.

catkira, 2023-01-31:
the compile script runs for about 5 minutes, using ghdl directly like shown in my last post only takes 3 seconds. I think the script needs some rework.
I call the script like this:
```
compile-xilinx-vivado.sh --unisim --vhdl93 --source /tools/Xilinx/Vivado/2022.2/data/vhdl/src/
```

catkira, 2023-01-31:
I am also a bit confused how the `-P` option works. 
To get my code running, I have to use `-Punisim` with `unisim-obj93.cf` in my src code folder. But the GHDL manual says for `-P`:
`Add DIRECTORY to the end of the list of directories to be searched for library files`

Paebbels, 2023-02-01:
A rund time of 3 seconds it too short :).

For the documentation about `-P` have a look here: https://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-P
> A library is searched in `DIRECTORY` and also in `DIRECTORY`/`LIB`/v`VV` (where `LIB` is the name of the library and `VV` the VHDL standard).

The second sentence is important too.

So use this:
`ghdl .... -Pxilinx ...`
and it will search the `unisim` library in `./xilinx/unisim/v93/`.