ghdl/ghdl #2335: Wrong driver on partial assignment of composite signal in process

simon-codes-something, 2023-01-30:
**Description**
I like to generalize my designs and therefore grouped multiple stream busses into a single signal.

However, GHDL currently thinks there is a multiply driven signal (`ghdl:error: several sources for unresolved signal`) when the grouped signal has both
- a part driven directly in the architecture
- a different part driven by a loop in a process
Manually unrolling the loop resolves the error message.

This problem has been reported before in https://github.com/ghdl/ghdl/issues/1135, which was closed with the explanation _A driver is added for the longest static prefixes._ (https://github.com/ghdl/ghdl/issues/1135#issuecomment-585341132)
However i think the LRM got misinterpreted a bit:

**Expected behaviour**
According to the LRM _There is a single driver for a given scalar signal S in a process statement, provided that there is at least one signal assignment statement in that process statement and that the longest static prefix of the target signal of that signal assignment statement denotes S or denotes a composite signal of which S is a subelement._ (LRM 1993, 2008 and 2019)

_There is a single driver for a given scalar signal S in a process statement, provided that [...]_ indicates that only scalar signals **S** get a driver. The rest of the sentence are the conditions under which this happens and basically means that **S** gets driven if either
- _the longest static prefix [...] denotes S_
- _the longest static prefix [...] denotes a composite signal of which S is a subelement_
 
The current implementation seems fine for the first case.

In the second case, GHDL creates a driver on the longest static prefix (the composite signal), while the LRM mentions a driver on S (the scalar subelement of that).

So, GHDL should instead create a single driver for each assigned scalar subelement being assigned in the process.
For example, in
```vhd 
for portname in tStreamName loop
     sDataStreams(portname).tReady <= true;
end loop;
```
a driver should be created only for `sDataStreams(A).tReady` and `sDataStreams(B).tReady` (instead of the currently implemented combined driver for `sDataStreams`).

**How to reproduce?**

```vhd :file: ent.vhd
library IEEE;
use IEEE.std_logic_1164.all;

entity multiplyDrivenBug_tb is
end entity multiplyDrivenBug_tb;

architecture myDescriptiveArchitectureName of multiplyDrivenBug_tb is
    type tStreamName is (A, B);
    type tStream is record
        tData : natural;
        tValid: boolean;
        tReady: boolean;
    end record;
    type tStreams is array(tStreamName) of tStream;

    signal clk          : std_logic := '0';
    signal sDataStreams : tStreams;
begin
    clk <= not clk after 5 ns;

    sDataStreams(A).tData <= 17;
    sDataStreams(A).tValid <= true;

    sDataStreams(B).tData <= 42;
    sDataStreams(B).tValid <= true;

    process(clk)
    begin
        if rising_edge(clk) then
            -- (A) works as intended
            -- sDataStreams(A).tReady <= true;
            -- sDataStreams(B).tReady <= true;

            -- (B) is more generic (and should otherwise do the same)
            -- ... but ghdl throws "several sources for unresolved signal"
            for portname in tStreamName loop
                sDataStreams(portname).tReady <= true;
            end loop;
        end if;
    end process;
end architecture myDescriptiveArchitectureName;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run multiplyDrivenBug_tb
```
**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [x] Released binaries: `[tarball_url](https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-20.04-mcode.tgz)`
  - [ ] Built from sources: `commit SHA`

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2023-01-30:
I am pretty sure you are not correct.

So, yes a driver should be created for sDataStreams(A).tReady and sDataStreams(B).tReady.  No question about that.
But what about sDataStreams(A).tValid ?  In the process, there is a signal assignment and the longest static prefix of the target signal of that assignment assignment denotes a composite signal of which S is a subelement.
Therefore a driver should be also created for sDataStreams(A).tValid, hence the presence of several sources.

Does it sound correct ?

simon-codes-something, 2023-01-30:
`sDataStreams(A).tValid` is not part of the signal assignment in the process, so would it really count as being _given_ _in a process statement_?

tgingold, 2023-01-30:
I would say so, because if you read LRM2008 14.7.2, there would be a driver for sDataStreams(A).tValid

As far as I know, the other simulators agree with that.

simon-codes-something, 2023-01-30:
hmm... not really happy with that reverse deduction. On the other hand after thinking about it a bit more i guess you are right as the part with the _longest static prefix_ would be superfluous otherwise.

I guess i will have to transfer all my signal assignments into the same process as a workaround. Not nice, but could be worse ;-)

closed 2023-01-30 by simon-codes-something

tgingold, 2023-01-30:
This rule about drivers makes writing code with records and arrays not that easy.
Maybe the way you propose it would be better than the current rule, but hard to change the rule.