ghdl/ghdl #2336: PSL: next_e operator misbehaviour

tmeissner, 2023-01-30:
**Description**

This seems to be a regression, as I think, the example worked in the past. But I'm not sure.

I've stumbled over some misbehaviour of the PSL `next_e` operator. The form `next e[i to j](a)` indicates that `a` should hold on at least one of the next `i`'th through `j`'th cycles. However, the implementation in GHDL only works if `a` holds on the last cycle of the given range.

For example:

```shell
   012345678901
a: __-_________
b: _____-______
c: ______-_____
d: _______-____
```
b, c and d each hold on one of the 3 cycles given in the range [3:5] .

```vhdl
NEXT_0_a : assert always (a -> next_e[3 to 5] (b));
NEXT_1_a : assert always (a -> next_e[3 to 5] (c));
NEXT_2_a : assert always (a -> next_e[3 to 5] (d));
```
All these 3 assertions should hold, but the first 2 are failing.

**Expected behaviour**
Assertions `NEXT_0_a` and `NEXT_1_a`  should also hold.

**How to reproduce?**
Get this file: https://github.com/tmeissner/psl_with_ghdl/blob/master/issues/issue_233x.vhd

```sh :image: ghdl/vunit:llvm
ghdl -a --std=08 issue_233x.vhd
ghdl -e --std=08 test_issue
```

If you run it:

```sh :image: hdlc/formal:all
$ ghdl -r --std=08 test_issue
issue_233x.vhd:97:3:@8ns:(psl assertion error): NEXT_4_a failed
issue_233x.vhd:93:3:@8ns:(psl assertion error): NEXT_3_a failed
issue_233x.vhd:85:3:@8ns:(psl assertion error): NEXT_1_a failed
issue_233x.vhd:81:3:@8ns:(psl assertion error): NEXT_0_a failed
simulation stopped @10ns with status 0
```
The assertions  `NEXT_0_a` and `NEXT_1_a` should hold.

- OS: Debian Bullseye
- Origin:
  - [x] Docker (Image:hdlc/formal:all)

tgingold, 2023-01-31:
Yes, you're right.  I am testing a fix.

Thank you for the issue.

closed 2023-02-01 by tgingold

tmeissner, 2023-02-03:
Thanks for fixing this so quickly :smiley:

commit 514ed1a8 (2023-02-01): psl-build: fix fusion with epsilon.  Fix #2336
commit 647546bd (2023-02-03): vhdl-sem_decls: fix a regression on subtype indication.  Fix #2336